kelly/node-i2c #50: writeBytes method

wakatanka, 2015-01-14:
Hi, i'm trying to read temperature from a sensor (sht21) with a raspberry pi, here is my working code:

var i2c = require('i2c');
var address = 0x40;
var wire = new i2c(address, {device: '/dev/i2c-1',debug: false}); 

wire.scan(function(err, data) {
  // result contains an array of addresses
  if (err !== null) {
    console.log('error scan: ' + err);
  } else {
    console.log("scan data: "+data);
  }
});

wire.writeByte(0xF3, function(err) {
  if (err !== null) {
    console.log("error write: " + err);
  } else {
    console.log("write ok");
  }
});

wire.readByte(function(err, res) {
     if (err !== null) {
       console.log("error readByte: " + err);
     } else {
       console.log(res);
    }
})

this works flawless, but i need to read the first 3 bytes to get all data, so i will use readbytes method with this code:

wire.readBytes(0x00, 3, function(err, res) {
  // result contains a buffer of bytes
  //var value = res.readUInt8(0);
  if (err !== null) {
    console.log("readBytes: " + err);
  } else {
    console.log(res);
  }
});

but it gives me an error. Probably i didn't understand what i have to pass as command parameter.

kelly, 2015-01-16:
I think you want to do wire.readBytes(0xF3, 3, function ....
The initial write that you're doing is the command byte. That's telling your device that you want to receive 3 bytes of data.

closed 2015-01-16 by kelly

dsteinborn, 2015-02-08:
Hi @wakatanka I'm also using the sh21t sensor with a raspberry pi. Have a look at pull request #35 from @bbx10 which contains a fix to read bytes without sending the command/register parameter. This fix also works for sh21t sensor.

bbx10, 2015-02-10:
I closed pull request #35 because the same functionality is available using plain read. For example, wire.read(3, function...). My fork is obsolete since @kib357 added plain I2C read and write.