kelly/node-i2c #51: Read or write raw bytes

joraff, 2015-02-02:
Is it possible to read or write raw bytes to a i2c device using this library?

I'm trying to interface with the [SL030 RFID module](http://www.stronglink-rfid.com/download/SL030-User-Manual.pdf), and it has an atypical command structure.

Instead of the usual sequence of ```slave address, ack, command register, ack, data(n)*ack, stop``` for a write, this module wants ```slave address, ack, length, ack, command, ack, data(n)*ack, stop```

Tracing a writeBytes all the way to the ioctl structure, it seems that it's assuming a command register will always be the first byte. Anything else in there that I've overlooked?

kelly, 2015-02-08:
Version 0.1.7 has this functionality. There is now a read and write method that doesn't require a command byte.

closed 2015-02-08 by kelly

joraff, 2015-02-09:
:clap: 

Thanks. I'll test it out. I can take care of a PR to update the docs. That's easy enough