kelly/node-i2c #53: setImmediate is not defined

jurafa, 2015-02-18:
Hi, I am trying to use i2c library to communicate between raspberry pi and unipi and when I try to include it (var i2c = require('i2c');) I receive this error. Any idea what goes wrong?
I am installing i2c using npm.

ReferenceError: setImmediate is not defined
    at Object.<anonymous> (/usr/share/adafruit/webide/repositories/my-pi-projects/rpi_unipi/node_modules/i2c/lib/i2c.coffee:12:10)
    at Object.<anonymous> (/usr/share/adafruit/webide/repositories/my-pi-projects/rpi_unipi/node_modules/i2c/lib/i2c.coffee:164:4)
    at Module._compile (module.js:446:26)
    at Object..coffee (/usr/share/adafruit/webide/repositories/my-pi-projects/rpi_unipi/node_modules/i2c/node_modules/coffee-script/lib/coffee-script/coffee-script.js:22:21)
    at Module.load (module.js:353:32)
    at Function._load (module.js:311:12)
    at Module.require (module.js:359:17)
    at require (module.js:375:17)
    at Object.<anonymous> (/usr/share/adafruit/webide/repositories/my-pi-projects/rpi_unipi/node_modules/i2c/main.js:2:11)
    at Module._compile (module.js:446:26)

joraff, 2015-02-18:
setImmediate was introduced in node v0.10. What version of node are you using? (Raspbian's repos contain a *really* old version, 0.6 I believe)

fivdi, 2015-02-18:
Does it work if the following line ([from here](https://github.com/kelly/node-i2c/blob/master/lib/i2c.coffee#L4)):
```coffeescript
 tick = setImmediate || process.nextTick
```
is changed to:
```coffeescript
 tick = global.setImmediate || process.nextTick
```

jurafa, 2015-02-20:
Thanks for help. The version of node on repository is 0.6.x. So I updated node and all works fine.

closed 2015-02-20 by jurafa