kelly/node-i2c #54: Can't install on Raspberry Pi 2 Model B

bishopmatthew, 2015-02-20:
Any idea what's going wrong?

`npm install i2c`

contents of npm-debug.log

```
0 info it worked if it ends with ok
1 verbose cli [ '/usr/local/bin/node', '/usr/local/bin/npm', 'install', 'i2c' ]
2 info using npm@2.5.1
3 info using node@v0.12.0
4 verbose config Skipping project config: /home/pi/.npmrc. (matches userconfig)
5 silly cache add args [ 'i2c', null ]
6 verbose cache add spec i2c
7 silly cache add parsed spec { raw: 'i2c',
7 silly cache add   scope: null,
7 silly cache add   name: 'i2c',
7 silly cache add   rawSpec: '',
7 silly cache add   spec: '*',
7 silly cache add   type: 'range' }
8 verbose addNamed i2c@*
9 silly addNamed semver.valid null
10 silly addNamed semver.validRange *
11 silly addNameRange { name: 'i2c', range: '*', hasData: false }
12 silly mapToRegistry name i2c
13 silly mapToRegistry using default registry
14 silly mapToRegistry registry https://registry.npmjs.org/
15 silly mapToRegistry uri https://registry.npmjs.org/i2c
16 verbose addNameRange registry:https://registry.npmjs.org/i2c not in flight; fetching
17 verbose request uri https://registry.npmjs.org/i2c
18 verbose request no auth needed
19 info attempt registry request try #1 at 00:33:45
20 verbose request id d0924a9cf04d4015
21 verbose etag "DC2IZQMDY6QKT6HBVFA1CEFIH"
22 http request GET https://registry.npmjs.org/i2c
23 http 304 https://registry.npmjs.org/i2c
24 silly get cb [ 304,
24 silly get   { date: 'Fri, 20 Feb 2015 00:33:45 GMT',
24 silly get     server: 'Apache',
24 silly get     via: '1.1 varnish',
24 silly get     'last-modified': 'Fri, 20 Feb 2015 00:33:45 GMT',
24 silly get     'cache-control': 'max-age=60',
24 silly get     etag: '"DC2IZQMDY6QKT6HBVFA1CEFIH"',
24 silly get     age: '0',
24 silly get     'x-served-by': 'cache-lax1427-LAX',
24 silly get     'x-cache': 'MISS',
24 silly get     'x-cache-hits': '0',
24 silly get     'x-timer': 'S1424392425.464737,VS0,VE180',
24 silly get     vary: 'Accept',
24 silly get     'content-length': '0',
24 silly get     'keep-alive': 'timeout=10, max=50',
24 silly get     connection: 'Keep-Alive' } ]
25 verbose etag https://registry.npmjs.org/i2c from cache
26 verbose get saving i2c to /home/pi/.npm/registry.npmjs.org/i2c/.cache.json
27 silly addNameRange number 2 { name: 'i2c', range: '*', hasData: true }
28 silly addNameRange versions [ 'i2c',
28 silly addNameRange   [ '0.0.1',
28 silly addNameRange     '0.0.2',
28 silly addNameRange     '0.0.3',
28 silly addNameRange     '0.0.4',
28 silly addNameRange     '0.0.5',
28 silly addNameRange     '0.1.0',
28 silly addNameRange     '0.1.1',
28 silly addNameRange     '0.1.2',
28 silly addNameRange     '0.1.3',
28 silly addNameRange     '0.1.4',
28 silly addNameRange     '0.1.5',
28 silly addNameRange     '0.1.6',
28 silly addNameRange     '0.1.7',
28 silly addNameRange     '0.1.8' ] ]
[... 536 lines cut ...]
319 error Linux 3.18.7-v7+
320 error argv "/usr/local/bin/node" "/usr/local/bin/npm" "install" "i2c"
321 error node v0.12.0
322 error npm  v2.5.1
323 error code ELIFECYCLE
324 error i2c@0.1.8 install: `node-gyp rebuild`
324 error Exit status 1
325 error Failed at the i2c@0.1.8 install script 'node-gyp rebuild'.
325 error This is most likely a problem with the i2c package,
325 error not with npm itself.
325 error Tell the author that this fails on your system:
325 error     node-gyp rebuild
325 error You can get their info via:
325 error     npm owner ls i2c
325 error There is likely additional logging output above.
326 verbose exit [ 1, true ]
327 verbose unbuild node_modules/i2c
328 info preuninstall i2c@0.1.8
329 info uninstall i2c@0.1.8
330 verbose unbuild rmStuff i2c@0.1.8 from /home/pi/node_modules
331 info postuninstall i2c@0.1.8
332 verbose gentlyRm vacuuming /home/pi/node_modules/i2c
333 silly gentlyRm purging /home/pi/node_modules/i2c
334 silly gentlyRm quitting because other entries in /home/pi/node_modules
```

joraff, 2015-02-20:
Is this on raspbian? do you have the build-essential package installed?

bishopmatthew, 2015-02-20:
Yes, and yes.
`dpkg -s build-essential`
```
Package: build-essential
Status: install ok installed
Priority: optional
Section: devel
Installed-Size: 36
Maintainer: Matthias Klose <doko@debian.org>
Architecture: armhf
Source: build-essential (11.5)
Version: 11.5+b1
Depends: libc6-dev | libc-dev, gcc (>= 4:4.4.3), g++ (>= 4:4.4.3), make, dpkg-dev (>= 1.13.5)
Description: Informational list of build-essential packages
 If you do not plan to build Debian packages, you don't need this
 package.  Starting with dpkg (>= 1.14.18) this package is required
 for building Debian packages.
 .
 This package contains an informational list of packages which are
 considered essential for building Debian packages.  This package also
 depends on the packages on that list, to make it easy to have the
 build-essential packages installed.
 .
 If you have this package installed, you only need to install whatever
 a package specifies as its build-time dependencies to build the
 package.  Conversely, if you are determining what your package needs
 to build-depend on, you can always leave out the packages this
 package depends on.
 .
 This package is NOT the definition of what packages are
 build-essential; the real definition is in the Debian Policy Manual.
 This package contains merely an informational list, which is all
 most people need.   However, if this package and the manual disagree,
 the manual is correct.
```

joraff, 2015-02-20:
Yeah, figured but wanted to check. I think the message is clearer if it's missing.

I'm using the same distro and kernel version, but with node v0.10.

bishopmatthew, 2015-02-20:
Ok, downgrading worked. Thanks!

closed 2015-02-20 by bishopmatthew

joraff, 2015-02-20:
Cool! Tho, an issue for v0.12 compatibility that someone can close via a PR isn't a bad idea to have around.

kelly, 2015-03-03:
0.12 is now supported in the current release thanks to @emersion

billtrudell, 2017-05-02:
This seem v7.8.0s like a house of cards, I'm trying to use the latest tools, node 7.8.0; but I'd have to go back version 0.12 to install a package that essentially doesn't seem to be supported.  I understand that if I want it, I could upgrade, but the whole idea of using open-source shared components is that if they are published, they should work.  I've wasted many hours with install errors and dependencies that are out-of-date.  It's very frustrating.  I shouldn't have to downgrade to years-old versions to install a package only to upgrade back to the latest package of node.  How do I know what those old dependencies are and if the code even works any more.

kelly, 2017-05-02:
@billtrudell dependency management is one of the hardest problems to solve, but it's not really related to this module or this thread.