kelly/node-i2c #55: Error installing on node 0.12.0

abiertob, 2015-02-22:
Trying to install i2c on raspberry pi with latest raspbian (3.18 kernel) and nodejs version 0.12.0 results in the following error:

<....cut...>

i2c.target.mk:81: recipe for target 'Release/obj.target/i2c/src/i2c.o' failed
make: *** [Release/obj.target/i2c/src/i2c.o] Error 1
make: Leaving directory '/home/pi/i2c/node_modules/i2c/build'
gyp ERR! build error
gyp ERR! stack Error: `make` failed with exit code: 2
gyp ERR! stack     at ChildProcess.onExit (/usr/local/lib/node_modules/npm/node_modules/node-gyp/lib/build.js:267:23)
gyp ERR! stack     at ChildProcess.emit (events.js:110:17)
gyp ERR! stack     at Process.ChildProcess._handle.onexit (child_process.js:1067:12)
gyp ERR! System Linux 3.18.7-v7+
gyp ERR! command "node" "/usr/local/lib/node_modules/npm/node_modules/node-gyp/bin/node-gyp.js" "rebuild"
gyp ERR! cwd /home/pi/i2c/node_modules/i2c
gyp ERR! node -v v0.12.0
gyp ERR! node-gyp -v v1.0.2
gyp ERR! not ok
npm ERR! Linux 3.18.7-v7+
npm ERR! argv "/usr/local/bin/node" "/usr/local/bin/npm" "install" "i2c"
npm ERR! node v0.12.0
npm ERR! npm  v2.5.1
npm ERR! code ELIFECYCLE

npm ERR! i2c@0.1.8 install: `node-gyp rebuild`
npm ERR! Exit status 1
npm ERR!
npm ERR! Failed at the i2c@0.1.8 install script 'node-gyp rebuild'.
npm ERR! This is most likely a problem with the i2c package,
npm ERR! not with npm itself.
npm ERR! Tell the author that this fails on your system:
npm ERR!     node-gyp rebuild
npm ERR! You can get their info via:
npm ERR!     npm owner ls i2c
npm ERR! There is likely additional logging output above.

npm ERR! Please include the following file with any support request:
npm ERR!     /home/pi/i2c/npm-debug.log

Installed on Raspberry pi v2
npm  v2.5.1
nodejs v0.12.0

Most likely it is because the library is not compatible with the latest nodejs version , or is there anything I am not seeing?

Thanks!

kelly, 2015-02-22:
Not compatible with the latest version of node.

closed 2015-02-22 by kelly

kelly, 2015-03-03:
0.12 is now supported in the current release thanks to @emersion

abiertob, 2015-03-04:
Thanks for letting me know!