kelly/node-i2c #56: Building on Node 0.12

emersion, 2015-03-02:
```
pi@blueberrypi ~/node $ npm install i2c --save
/
> i2c@0.1.8 install /home/pi/node/node_modules/i2c
> node-gyp rebuild

child_process: customFds option is deprecated, use stdio instead.
make: Entering directory '/home/pi/node/node_modules/i2c/build'
  CXX(target) Release/obj.target/i2c/src/i2c.o
../src/i2c.cc: In function ‘void setAddress(int8_t)’:
../src/i2c.cc:21:28: error: ‘New’ is not a member of ‘v8::String’
../src/i2c.cc:22:5: error: ‘ThrowException’ was not declared in this scope
../src/i2c.cc: At global scope:
../src/i2c.cc:26:32: error: ‘Arguments’ does not name a type
../src/i2c.cc:26:43: error: ISO C++ forbids declaration of ‘args’ with no type [-fpermissive]
/home/pi/.node-gyp/0.12.0/deps/v8/include/v8.h: In function ‘v8::Handle<v8::Value> SetAddress(const int&)’:
/home/pi/.node-gyp/0.12.0/deps/v8/include/v8.h:816:13: error: ‘v8::HandleScope::HandleScope()’ is protected
../src/i2c.cc:27:15: error: within this context
../src/i2c.cc:29:16: error: invalid types ‘const int[int]’ for array subscript
../src/i2c.cc:32:16: error: ‘class v8::HandleScope’ has no member named ‘Close’
../src/i2c.cc:32:32: error: too few arguments to function ‘v8::Handle<v8::Primitive> v8::Undefined(v8::Isolate*)’
/home/pi/.node-gyp/0.12.0/deps/v8/include/v8.h:305:28: note: declared here
../src/i2c.cc: At global scope:
../src/i2c.cc:35:26: error: ‘Arguments’ does not name a type
../src/i2c.cc:35:37: error: ISO C++ forbids declaration of ‘args’ with no type [-fpermissive]
/home/pi/.node-gyp/0.12.0/deps/v8/include/v8.h: In function ‘v8::Handle<v8::Value> Scan(const int&)’:
/home/pi/.node-gyp/0.12.0/deps/v8/include/v8.h:816:13: error: ‘v8::HandleScope::HandleScope()’ is protected
../src/i2c.cc:36:15: error: within this context
../src/i2c.cc:39:58: error: invalid types ‘const int[int]’ for array subscript
../src/i2c.cc:40:38: error: invalid conversion from ‘int’ to ‘v8::Isolate*’ [-fpermissive]
/home/pi/.node-gyp/0.12.0/deps/v8/include/v8.h:2417:23: error:   initializing argument 1 of ‘static v8::Local<v8::Array> v8::Array::New(v8::Isolate*, int)’ [-fpermissive]
../src/i2c.cc:41:45: error: too few arguments to function ‘v8::Handle<v8::Primitive> v8::Null(v8::Isolate*)’
/home/pi/.node-gyp/0.12.0/deps/v8/include/v8.h:306:28: note: declared here
../src/i2c.cc:53:37: error: no matching function for call to ‘v8::Integer::New(int&)’
../src/i2c.cc:53:37: note: candidate is:
/home/pi/.node-gyp/0.12.0/deps/v8/include/v8.h:2012:25: note: static v8::Local<v8::Integer> v8::Integer::New(v8::Isolate*, int32_t)
/home/pi/.node-gyp/0.12.0/deps/v8/include/v8.h:2012:25: note:   candidate expects 2 arguments, 1 provided
../src/i2c.cc:60:18: error: ‘GetCurrent’ is not a member of ‘v8::Context’
../src/i2c.cc:62:16: error: ‘class v8::HandleScope’ has no member named ‘Close’
../src/i2c.cc: At global scope:
../src/i2c.cc:65:27: error: ‘Arguments’ does not name a type
../src/i2c.cc:65:38: error: ISO C++ forbids declaration of ‘args’ with no type [-fpermissive]
/home/pi/.node-gyp/0.12.0/deps/v8/include/v8.h: In function ‘v8::Handle<v8::Value> Close(const int&)’:
/home/pi/.node-gyp/0.12.0/deps/v8/include/v8.h:816:13: error: ‘v8::HandleScope::HandleScope()’ is protected
../src/i2c.cc:66:15: error: within this context
../src/i2c.cc:71:16: error: ‘class v8::HandleScope’ has no member named ‘Close’
../src/i2c.cc:71:32: error: too few arguments to function ‘v8::Handle<v8::Primitive> v8::Undefined(v8::Isolate*)’
/home/pi/.node-gyp/0.12.0/deps/v8/include/v8.h:305:28: note: declared here
../src/i2c.cc: At global scope:
../src/i2c.cc:74:26: error: ‘Arguments’ does not name a type
../src/i2c.cc:74:37: error: ISO C++ forbids declaration of ‘args’ with no type [-fpermissive]
/home/pi/.node-gyp/0.12.0/deps/v8/include/v8.h: In function ‘v8::Handle<v8::Value> Open(const int&)’:
/home/pi/.node-gyp/0.12.0/deps/v8/include/v8.h:816:13: error: ‘v8::HandleScope::HandleScope()’ is protected
../src/i2c.cc:75:15: error: within this context
../src/i2c.cc:77:34: error: invalid types ‘const int[int]’ for array subscript
../src/i2c.cc:79:45: error: too few arguments to function ‘v8::Handle<v8::Primitive> v8::Null(v8::Isolate*)’
/home/pi/.node-gyp/0.12.0/deps/v8/include/v8.h:306:28: note: declared here
../src/i2c.cc:83:28: error: ‘New’ is not a member of ‘v8::String’
../src/i2c.cc:86:13: error: invalid types ‘const int[int]’ for array subscript
../src/i2c.cc:88:60: error: invalid types ‘const int[int]’ for array subscript
../src/i2c.cc:91:20: error: ‘GetCurrent’ is not a member of ‘v8::Context’
../src/i2c.cc:94:16: error: ‘class v8::HandleScope’ has no member named ‘Close’
../src/i2c.cc:94:32: error: too few arguments to function ‘v8::Handle<v8::Primitive> v8::Undefined(v8::Isolate*)’
/home/pi/.node-gyp/0.12.0/deps/v8/include/v8.h:305:28: note: declared here
../src/i2c.cc: At global scope:
../src/i2c.cc:97:26: error: ‘Arguments’ does not name a type
../src/i2c.cc:97:37: error: ISO C++ forbids declaration of ‘args’ with no type [-fpermissive]
/home/pi/.node-gyp/0.12.0/deps/v8/include/v8.h: In function ‘v8::Handle<v8::Value> Read(const int&)’:
/home/pi/.node-gyp/0.12.0/deps/v8/include/v8.h:816:13: error: ‘v8::HandleScope::HandleScope()’ is protected
../src/i2c.cc:98:15: error: within this context
[... 213 lines cut ...]
gyp ERR! cwd /home/pi/node/node_modules/i2c
gyp ERR! node -v v0.12.0
gyp ERR! node-gyp -v v1.0.2
gyp ERR! not ok 
npm ERR! Linux 3.18.5+
npm ERR! argv "/usr/local/bin/node" "/usr/local/bin/npm" "install" "i2c" "--save"
npm ERR! node v0.12.0
npm ERR! npm  v2.5.1
npm ERR! code ELIFECYCLE

npm ERR! i2c@0.1.8 install: `node-gyp rebuild`
npm ERR! Exit status 1
npm ERR! 
npm ERR! Failed at the i2c@0.1.8 install script 'node-gyp rebuild'.
npm ERR! This is most likely a problem with the i2c package,
npm ERR! not with npm itself.
npm ERR! Tell the author that this fails on your system:
npm ERR!     node-gyp rebuild
npm ERR! You can get their info via:
npm ERR!     npm owner ls i2c
npm ERR! There is likely additional logging output above.

npm ERR! Please include the following file with any support request:
npm ERR!     /home/pi/node/npm-debug.log
```

kelly, 2015-03-02:
Not supported at this time.

emersion, 2015-03-02:
Can I make a pull request?

kelly, 2015-03-02:
Sure, as long as it looks good, i'll merge it in.

closed 2015-03-03 by kelly