kelly/node-i2c #60: Require slow due to coffee (v0.1.8)

gfduszynski, 2015-06-01:
I'm using CORE9G25@400MHz module with node-i2c.
Require call to node-i2c (v0.1.8) takes about ~4100ms, 
out of which nested require to coffee-script takes about ~1900ms and compiling also about 2s.

Switching to JS would provide massive reduction in startup time.

kelly, 2015-06-01:
It's a single coffeescript file, so just run `coffee -c i2c.coffee`

closed 2015-06-01 by kelly

gfduszynski, 2015-06-01:
This solution still leaves the dependency to 'coffee-script'.
Plus it's a thing that no one would like to have in a build job :smile:

kelly, 2015-06-01:
You can fork it too. Most people using this module aren't running it on a 400MHZ cpu. If that changes, I'll look into modifying the code.

gfduszynski, 2015-06-09:
Sure, most people aren't using this on their i7 either.
Anyway, node-i2c is a great lib, I'm just trying to get best possible startup time from this tiny CPU.
For now I've switched to i2c-bus. It's not so convenient but it works and loads under ~180ms.