kelly/node-i2c #62: Edison occasionally fails to write over the i2c-6 bus

mikealbinson, 2015-07-17:
I have had some issues when I boot up my intel edison after shutdown. For a period of time the error callback for i2c.writeBytes will return "unable to write" in regard to the i2c-6 bus with no other indication of what needs to be fixed. However, after this period of time the error will disappear and the i2c bus will work normally. Not sure if you've encountered this before/even intended this for use on the Edison, but the majority of the time it works wonderfully, so thank you very much for all of your hard work! Just thought I'd bring this to your attention.

kelly, 2015-07-24:
I'm not entirely sure what is happening here. Unfortunately the Edison isn't currently supported since I don't have access to a device.

mikealbinson, 2015-07-28:
Bummer, in the meantime, if anyone is experiencing the same issue I would recommend the node library mraa for the Intel Edison. It took a little work to switch over, but my code now runs consistently without the above issue.