kelly/node-i2c #66: Install fails on BBB w/Debian

stormbytes, 2015-08-21:
→ npm install i2c
-
> i2c@0.2.1 install /home/storm/node_modules/i2c
> node-gyp rebuild

make: Entering directory `/home/storm/node_modules/i2c/build'
  CXX(target) Release/obj.target/i2c/src/i2c.o
../src/i2c.cc: In function ‘void setAddress(int8_t)’:
../src/i2c.cc:22:14: error: ‘class v8::Isolate’ has no member named ‘ThrowException’
../src/i2c.cc:23:28: error: ‘NewFromUtf8’ is not a member of ‘v8::String’
../src/i2c.cc: At global scope:
../src/i2c.cc:29:23: error: ‘FunctionCallbackInfo’ does not name a type
../src/i2c.cc:29:23: error: ISO C++ forbids declaration of ‘parameter’ with no type [-fpermissive]
../src/i2c.cc:29:43: error: expected ‘,’ or ‘...’ before ‘<’ token
../src/i2c.cc: In function ‘void SetAddress(int)’:
../src/i2c.cc:31:28: error: no matching function for call to ‘v8::HandleScope::HandleScope(v8::Isolate*&)’
../src/i2c.cc:31:28: note: candidates are:
/usr/include/nodejs/deps/v8/include/v8.h:473:3: note: v8::HandleScope::HandleScope(const v8::HandleScope&)
/usr/include/nodejs/deps/v8/include/v8.h:473:3: note:   no known conversion for argument 1 from ‘v8::Isolate*’ to ‘const v8::HandleScope&’
/usr/include/nodejs/deps/v8/include/v8.h:448:3: note: v8::HandleScope::HandleScope()
/usr/include/nodejs/deps/v8/include/v8.h:448:3: note:   candidate expects 0 arguments, 1 provided
../src/i2c.cc:33:10: error: ‘args’ was not declared in this scope
../src/i2c.cc: At global scope:
../src/i2c.cc:37:17: error: ‘FunctionCallbackInfo’ does not name a type
../src/i2c.cc:37:17: error: ISO C++ forbids declaration of ‘parameter’ with no type [-fpermissive]
../src/i2c.cc:37:37: error: expected ‘,’ or ‘...’ before ‘<’ token
../src/i2c.cc: In function ‘void Scan(int)’:
../src/i2c.cc:39:28: error: no matching function for call to ‘v8::HandleScope::HandleScope(v8::Isolate*&)’
../src/i2c.cc:39:28: note: candidates are:
/usr/include/nodejs/deps/v8/include/v8.h:473:3: note: v8::HandleScope::HandleScope(const v8::HandleScope&)
/usr/include/nodejs/deps/v8/include/v8.h:473:3: note:   no known conversion for argument 1 from ‘v8::Isolate*’ to ‘const v8::HandleScope&’
/usr/include/nodejs/deps/v8/include/v8.h:448:3: note: v8::HandleScope::HandleScope()
/usr/include/nodejs/deps/v8/include/v8.h:448:3: note:   candidate expects 0 arguments, 1 provided
../src/i2c.cc:42:52: error: ‘args’ was not declared in this scope
../src/i2c.cc:43:49: error: no matching function for call to ‘v8::Array::New(v8::Isolate*&, int)’
../src/i2c.cc:43:49: note: candidate is:
/usr/include/nodejs/deps/v8/include/v8.h:1758:32: note: static v8::Local<v8::Array> v8::Array::New(int)
/usr/include/nodejs/deps/v8/include/v8.h:1758:32: note:   candidate expects 1 argument, 2 provided
../src/i2c.cc:44:62: error: no matching function for call to ‘v8::Local<v8::Value>::New(v8::Isolate*&, v8::Handle<v8::Primitive>)’
../src/i2c.cc:44:62: note: candidate is:
/usr/include/nodejs/deps/v8/include/v8.h:4197:10: note: static v8::Local<T> v8::Local<T>::New(v8::Handle<T>) [with T = v8::Value]
/usr/include/nodejs/deps/v8/include/v8.h:4197:10: note:   candidate expects 1 argument, 2 provided
../src/i2c.cc:56:46: error: invalid conversion from ‘v8::Isolate*’ to ‘int32_t {aka int}’ [-fpermissive]
/usr/include/nodejs/deps/v8/include/v8.h:1392:34: error:   initializing argument 1 of ‘static v8::Local<v8::Integer> v8::Integer::New(int32_t, v8::Isolate*)’ [-fpermissive]
../src/i2c.cc:56:46: error: invalid conversion from ‘int’ to ‘v8::Isolate*’ [-fpermissive]
/usr/include/nodejs/deps/v8/include/v8.h:1392:34: error:   initializing argument 2 of ‘static v8::Local<v8::Integer> v8::Integer::New(int32_t, v8::Isolate*)’ [-fpermissive]
../src/i2c.cc:63:27: error: ‘class v8::Isolate’ has no member named ‘GetCurrentContext’
../src/i2c.cc: At global scope:
../src/i2c.cc:68:18: error: ‘FunctionCallbackInfo’ does not name a type
../src/i2c.cc:68:18: error: ISO C++ forbids declaration of ‘parameter’ with no type [-fpermissive]
../src/i2c.cc:68:38: error: expected ‘,’ or ‘...’ before ‘<’ token
../src/i2c.cc: In function ‘void Close(int)’:
../src/i2c.cc:70:28: error: no matching function for call to ‘v8::HandleScope::HandleScope(v8::Isolate*&)’
../src/i2c.cc:70:28: note: candidates are:
/usr/include/nodejs/deps/v8/include/v8.h:473:3: note: v8::HandleScope::HandleScope(const v8::HandleScope&)
/usr/include/nodejs/deps/v8/include/v8.h:473:3: note:   no known conversion for argument 1 from ‘v8::Isolate*’ to ‘const v8::HandleScope&’
/usr/include/nodejs/deps/v8/include/v8.h:448:3: note: v8::HandleScope::HandleScope()
/usr/include/nodejs/deps/v8/include/v8.h:448:3: note:   candidate expects 0 arguments, 1 provided
../src/i2c.cc: At global scope:
../src/i2c.cc:77:17: error: ‘FunctionCallbackInfo’ does not name a type
../src/i2c.cc:77:17: error: ISO C++ forbids declaration of ‘parameter’ with no type [-fpermissive]
../src/i2c.cc:77:37: error: expected ‘,’ or ‘...’ before ‘<’ token
../src/i2c.cc: In function ‘void Open(int)’:
../src/i2c.cc:79:28: error: no matching function for call to ‘v8::HandleScope::HandleScope(v8::Isolate*&)’
../src/i2c.cc:79:28: note: candidates are:
/usr/include/nodejs/deps/v8/include/v8.h:473:3: note: v8::HandleScope::HandleScope(const v8::HandleScope&)
/usr/include/nodejs/deps/v8/include/v8.h:473:3: note:   no known conversion for argument 1 from ‘v8::Isolate*’ to ‘const v8::HandleScope&’
/usr/include/nodejs/deps/v8/include/v8.h:448:3: note: v8::HandleScope::HandleScope()
/usr/include/nodejs/deps/v8/include/v8.h:448:3: note:   candidate expects 0 arguments, 1 provided
../src/i2c.cc:81:28: error: ‘args’ was not declared in this scope
[... 201 lines cut ...]
npm WARN For further explanations, please read
/usr/share/doc/nodejs/README.Debian

npm ERR! i2c@0.2.1 install: `node-gyp rebuild`
npm ERR! Exit status 1
npm ERR!
npm ERR! Failed at the i2c@0.2.1 install script.
npm ERR! This is most likely a problem with the i2c package,
npm ERR! not with npm itself.
npm ERR! Tell the author that this fails on your system:
npm ERR!     node-gyp rebuild
npm ERR! You can get their info via:
npm ERR!     npm owner ls i2c
npm ERR! There is likely additional logging output above.

npm ERR! System Linux 3.8.13-bone74
npm ERR! command "/usr/bin/nodejs" "/usr/bin/npm" "install" "i2c"
npm ERR! cwd /home/storm
npm ERR! node -v v0.10.38
npm ERR! npm -v 1.4.21
npm ERR! code ELIFECYCLE
npm ERR!
npm ERR! Additional logging details can be found in:
npm ERR!     /home/storm/npm-debug.log
npm ERR! not ok code 0

kelly, 2015-08-21:
https://github.com/kelly/node-i2c#node-011-and-under

closed 2015-08-21 by kelly

stormbytes, 2015-08-21:
Thanks! 

My bad!