kelly/node-i2c #69: Possible compatibility problem with node 4.0?  Can't build on Raspberry Pi - Debian Jessie

JohnNavratil, 2015-09-14:
node -v ==> v4.0.0
node-gyp -v ==> v3.0.2
uname -a ==> Linux rpi0 4.1.6+ #810 PREEMPT Tue Aug 18 15:19:58 BST 2015 armv6l GNU/Linux

Many C++ compile errors.  I'll be happy to generate a clean npm debug log if someone can tell me how to attach it.
--------------------------------------------------------------------------

root@rpi0: /home/jnavratil/pmon$ npm -g install i2c    
|
> i2c@0.2.1 install /usr/local/lib/node_modules/i2c
> node-gyp rebuild

make: Entering directory '/usr/local/lib/node_modules/i2c/build'
  CXX(target) Release/obj.target/i2c/src/i2c.o
../src/i2c.cc: In function 'void ReadBlock(const v8::FunctionCallbackInfo<v8::Value>&)':
../src/i2c.cc:162:47: error: no matching function for call to 'New(int32_t&)'
   Local<Object> buffer = node::Buffer::New(len);
                                               ^
../src/i2c.cc:162:47: note: candidates are:
In file included from ../src/i2c.cc:2:0:
/root/.node-gyp/4.0.0/include/node/node_buffer.h:28:40: note: v8::MaybeLocal<v8::Object> node::Buffer::New(v8::Isolate*, size_t)
 NODE_EXTERN v8::MaybeLocal<v8::Object> New(v8::Isolate* isolate, size_t length);
                                        ^
/root/.node-gyp/4.0.0/include/node/node_buffer.h:28:40: note:   candidate expects 2 arguments, 1 provided
/root/.node-gyp/4.0.0/include/node/node_buffer.h:31:40: note: v8::MaybeLocal<v8::Object> node::Buffer::New(v8::Isolate*, v8::Local<v8::String>, node::encoding)
 NODE_EXTERN v8::MaybeLocal<v8::Object> New(v8::Isolate* isolate,
                                        ^
/root/.node-gyp/4.0.0/include/node/node_buffer.h:31:40: note:   candidate expects 3 arguments, 1 provided
/root/.node-gyp/4.0.0/include/node/node_buffer.h:36:40: note: v8::MaybeLocal<v8::Object> node::Buffer::New(v8::Isolate*, char*, size_t, node::Buffer::FreeCallback, void*)
 NODE_EXTERN v8::MaybeLocal<v8::Object> New(v8::Isolate* isolate,
                                        ^
/root/.node-gyp/4.0.0/include/node/node_buffer.h:36:40: note:   candidate expects 5 arguments, 1 provided
/root/.node-gyp/4.0.0/include/node/node_buffer.h:43:40: note: v8::MaybeLocal<v8::Object> node::Buffer::New(v8::Isolate*, char*, size_t)
 NODE_EXTERN v8::MaybeLocal<v8::Object> New(v8::Isolate* isolate,
                                        ^
/root/.node-gyp/4.0.0/include/node/node_buffer.h:43:40: note:   candidate expects 3 arguments, 1 provided
i2c.target.mk:86: recipe for target 'Release/obj.target/i2c/src/i2c.o' failed
make: *** [Release/obj.target/i2c/src/i2c.o] Error 1
make: Leaving directory '/usr/local/lib/node_modules/i2c/build'
gyp ERR! build error 
gyp ERR! stack Error: `make` failed with exit code: 2
gyp ERR! stack     at ChildProcess.onExit (/usr/local/lib/node_modules/npm/node_modules/node-gyp/lib/build.js:270:23)
gyp ERR! stack     at emitTwo (events.js:87:13)
gyp ERR! stack     at ChildProcess.emit (events.js:172:7)
gyp ERR! stack     at Process.ChildProcess._handle.onexit (internal/child_process.js:200:12)
gyp ERR! System Linux 4.1.6+
gyp ERR! command "/usr/local/bin/node" "/usr/local/lib/node_modules/npm/node_modules/node-gyp/bin/node-gyp.js" "rebuild"
gyp ERR! cwd /usr/local/lib/node_modules/i2c
gyp ERR! node -v v4.0.0
gyp ERR! node-gyp -v v3.0.1
gyp ERR! not ok 
npm ERR! Linux 4.1.6+
npm ERR! argv "/usr/local/bin/node" "/usr/local/bin/npm" "-g" "install" "i2c"
npm ERR! node v4.0.0
npm ERR! npm  v2.14.2
npm ERR! code ELIFECYCLE

npm ERR! i2c@0.2.1 install: `node-gyp rebuild`
npm ERR! Exit status 1
npm ERR! 
npm ERR! Failed at the i2c@0.2.1 install script 'node-gyp rebuild'.
npm ERR! This is most likely a problem with the i2c package,
npm ERR! not with npm itself.
npm ERR! Tell the author that this fails on your system:
npm ERR!     node-gyp rebuild
npm ERR! You can get their info via:
npm ERR!     npm owner ls i2c
npm ERR! There is likely additional logging output above.

npm ERR! Please include the following file with any support request:
npm ERR!     /home/jnavratil/pmon/npm-debug.log

cricri0464, 2015-10-02:
Same issue here ...

mitulsuthar, 2015-10-07:
I have the same issue of it not compiling on a BeagleBone with Debian and Node v4.0.0.0 .

pseudobeer, 2015-10-08:
Exactly the same issue here.

(as a dependency of cylon-raspi)

kelly, 2015-10-08:
node 4.0 requires a major re-write of native v8 modules. I likely won't be able to get to this for sometime. If someone wants to tackle it and send me a pull request, i'll merge it, and submit a release.

JohnNavratil, 2015-10-08:
Kelly,

Thanks for the heads up.  In theory I could do this (I've been a C programmer for years, I'm not a C++ maven), but my complete unfamiliarity with the code base makes my success unlikely.

I'm just building a Raspian system and am not married to Node 4.0.  Do you have a clue to which node version is compatible with i2c?  If not, I'll report back with my results as I try to find one.

joraff, 2015-10-08:
In the mean time, I'm using [i2c-bus](https://github.com/fivdi/i2c-bus) with node v4

Masterbonnie, 2015-10-12:
The Node version that was compatible with i2c is Node 0.10     Stable Version

kelly, 2015-10-12:
That is incorrect, it should work for v0.12.7 and everything before. So everything prior to the 4.0 release.

Masterbonnie, 2015-10-12:
Oh thanks :D did you how to fix this problem?
I2c is no working and I need to install cylon-raspy.

kelly, 2015-10-12:
For now just use nvm to install node 0.12.7 and you should be able to use cylon-raspi with i2c.

anvanza, 2015-10-26:
This pull request is working: https://github.com/kelly/node-i2c/pull/70/commits
You can implement it immediately using the following snippet in your packages.json.
"i2c": "https://github.com/polaris/node-i2c"

Kepro, 2015-11-14:
+1 need this :)

jamms, 2015-12-17:
Anvanza, I don't suppose there is a way you can show me how to do this, or point me in the direction of some generic instructions. Am still quite new to this!

TIA

jamms, 2015-12-18:
Anvanza, never mind, I didn't take your comment literally enough. Works great for lots of my sensors, Thanks for the heads-up!

jnovack, 2016-01-16:
If your application has a module which has `node-i2c` as a subpackage, rather than modify the module, just install the workaround version globally.

`npm install -g https://github.com/jnovack/node-i2c/archive/master.zip`

When you `npm install` your application, the module will see you already have it and use the local "fixed" copy.

JohnNavratil, 2016-01-19:
jnovack,

I reviewed yours and anvanza's files and it seems yours is trying to be a more universal solutions.  So I tried it, and received....

npm install -g https://github.com/jnovack/node-i2c/archive/master.zip
npm ERR! not a package /tmp/npm-19977-c99c1b5a/github.com/jnovack/node-i2c/archive/master.zip
npm ERR! Linux 4.1.13+
npm ERR! argv "/usr/local/bin/node" "/usr/local/bin/npm" "install" "-g" "https://github.com/jnovack/node-i2c/archive/master.zip"
npm ERR! node v4.2.4
npm ERR! npm  v2.14.12
npm ERR! path /tmp/npm-19977-c99c1b5a/unpack-9d46181a9b21/package.json
npm ERR! code ENOENT
npm ERR! errno -2
npm ERR! syscall open

npm ERR! enoent ENOENT: no such file or directory, open '/tmp/npm-19977-c99c1b5a/unpack-9d46181a9b21/package.json'
npm ERR! enoent This is most likely not a problem with npm itself
npm ERR! enoent and is related to npm not being able to find a file.
npm ERR! enoent

Any thoughts?

You see the versions of node and npm.  This is being installed on a Raspberry Pi running an updated Jessie.

JohnNavratil, 2016-01-19:
jnovack,

Flying completely by the seat of my pants, and as root (shivver!), I...

 wget "https://github.com/jnovack/node-i2c/archive/master.zip"
 unzip master.zip
 cd node-i2c-master/
 npm -g install .

Things seemed to work.  I've not done any testing yet.  Thoughts?

jnovack, 2016-01-21:
Sorry, I tried to make it easy, but stupid npm doesn't see it as a package.

New instructions: 

    git clone https://github.com/jnovack/node-i2c/
    cd node-i2c/
    npm install -g .

[You can see the changes I've made here.](https://github.com/kelly/node-i2c/compare/master...jnovack:master) All I've done is isolate node version v4 code as a quick and dirty work-around and set the version to be `0.2.1-alpha.1`.  This ensures that any update of the REAL package will overwrite my package.

nkolban, 2016-01-23:
+1 on this being a problem for my projects

kawashimaken, 2016-03-09:
I need to install i2c which is used as a submodule of cylon-raspi. I encountered the same issue, I am using node.js 4.X stable.
I know that the Node version that was compatible with i2c is Node 0.12.x Stable version as mentioned above, however I want  to use the latest stable node.js to run my MEAN stack web application, from where I can hook up my raspberry and do some automation thing from my web page. 
The node.js 4.X support will be highly appreciated, thank you for the great work.

alexellis, 2016-03-09:
@kawashimaken This thread has been running for months, you'd be better off forking the cylon-raspi repo and altering it to use a different i2c library. I remember being pointed in the direction of one when I came here with the same issue. I'll look up the name for you.

kawashimaken, 2016-03-10:
@alexellis Thank you for your info and your advice! I will definitely check out your repo! I hope I can make a smooth swap to i2c-bus.

closed 2016-06-15 by kelly