kelly/node-i2c #71: Sensor data doesn't update after set time

chintanp, 2015-12-02:
I am using a TI gas-gauge which is a battery sensor, as in, it reports battery voltage, current, temp etc. The sensor works fine, as I have a working C code. Trying to port it to nodejs. An initial run with node-i2c gives me the results I am hoping for. Although, I haven't been able to update the results continuously. 
I even tried modifying the accel.coffee example for my sensor, which now looks like this: https://gist.github.com/chintanp/c8e1ad362eb9d99c30c7 

As you can see in the code, it has a ` this.wire.read ... ` inside a ` setTimeout ` which should get me the sensor readings every 100 ms, or atleast this is what I am trying to achieve. What happens right now, is the code prints the sensor value correctly, once, and then just sits there without exiting, or re-printing etc. I have tried putting the `wire.read` code in a `while(true)` , `for(;;)` to create an infinite loop type call, but to no avail. This may not be a node-i2c issue, just my bad nodejs.

Anyway, how can I get this working ?

closed 2015-12-02 by kelly

chintanp, 2015-12-02:
Closed without a comment ? Hey, some love would go a long way.

kelly, 2015-12-02:
Project issues aren't for debugging you code. Please use stackoverflow or other services for code help.