kelly/node-i2c #73: Install Problems on Wheezy 7.9

AndrewDugenske, 2016-02-09:
It looks like the library is not playing well with Debian Wheezy version 7.9.
When I ran npm install i2c, I received the following at the end of the install:

gyp ERR! build error 
gyp ERR! stack Error: `make` failed with exit code: 2
gyp ERR! stack     at ChildProcess.onExit (/usr/share/node-gyp/lib/build.js:267:23)
gyp ERR! stack     at ChildProcess.emit (events.js:98:17)
gyp ERR! stack     at Process.ChildProcess._handle.onexit (child_process.js:820:12)
gyp ERR! System Linux 3.8.13-bone71.1
gyp ERR! command "nodejs" "/usr/bin/node-gyp" "rebuild"
gyp ERR! cwd /root/node_modules/i2c
gyp ERR! node -v v0.10.41
gyp ERR! node-gyp -v v0.12.2
gyp ERR! not ok 
npm WARN This failure might be due to the use of legacy binary "node"
npm WARN For further explanations, please read
/usr/share/doc/nodejs/README.Debian
 
npm ERR! i2c@0.2.1 install: `node-gyp rebuild`
npm ERR! Exit status 1
npm ERR! 
npm ERR! Failed at the i2c@0.2.1 install script.
npm ERR! This is most likely a problem with the i2c package,
npm ERR! not with npm itself.
npm ERR! Tell the author that this fails on your system:
npm ERR!     node-gyp rebuild
npm ERR! You can get their info via:
npm ERR!     npm owner ls i2c
npm ERR! There is likely additional logging output above.

npm ERR! System Linux 3.8.13-bone71.1
npm ERR! command "/usr/bin/nodejs" "/usr/bin/npm" "install" "i2c"
npm ERR! cwd /root
npm ERR! node -v v0.10.41
npm ERR! npm -v 1.4.21
npm ERR! code ELIFECYCLE
npm ERR! 
npm ERR! Additional logging details can be found in:
npm ERR!     /root/npm-debug.log
npm ERR! not ok code 0

fafamendes92, 2016-05-10:
The same happens in version 8.3

alphacharlie, 2016-06-12:
node 0.10 needs a lower version of i2c

Do -
`npm install i2c@0.1.8`

chuber70, 2017-01-26:
I cannot install the i2c node.  I have attached the debug file.  I did try what you suggested above and it seemed to install but there were no i2c nodes in node red.  The debug file was generated using the "npm install --unsafe-perm node-red-contrib-i2c" from the node red directory.

[npm-debug.txt](https://github.com/kelly/node-i2c/files/732304/npm-debug.txt)

AndrewDugenske, 2019-06-18:
Closing due to no action.

closed 2019-06-18 by AndrewDugenske