kelly/node-i2c #75: npm install i2c failed

goog, 2016-04-13:
pi@raspberrypi:~ $ sudo npm install -g --unsafe-perm i2c

> i2c@0.2.1 install /usr/local/lib/node_modules/i2c
> node-gyp rebuild

make: Entering directory '/usr/local/lib/node_modules/i2c/build'
  CXX(target) Release/obj.target/i2c/src/i2c.o
../src/i2c.cc: In function ‘void ReadBlock(const v8::FunctionCallbackInfo<v8::Value>&)’:
../src/i2c.cc:162:47: error: no matching function for call to ‘New(int32_t&)’
   Local<Object> buffer = node::Buffer::New(len);
                                               ^
../src/i2c.cc:162:47: note: candidates are:
In file included from ../src/i2c.cc:2:0:
/root/.node-gyp/4.2.1/include/node/node_buffer.h:28:40: note: v8::MaybeLocal<v8::Object> node::Buffer::New(v8::Isolate*, size_t)
 NODE_EXTERN v8::MaybeLocal<v8::Object> New(v8::Isolate* isolate, size_t length);
                                        ^
/root/.node-gyp/4.2.1/include/node/node_buffer.h:28:40: note:   candidate expects 2 arguments, 1 provided
/root/.node-gyp/4.2.1/include/node/node_buffer.h:31:40: note: v8::MaybeLocal<v8::Object> node::Buffer::New(v8::Isolate*, v8::Local<v8::String>, node::encoding)
 NODE_EXTERN v8::MaybeLocal<v8::Object> New(v8::Isolate* isolate,
                                        ^
/root/.node-gyp/4.2.1/include/node/node_buffer.h:31:40: note:   candidate expects 3 arguments, 1 provided
/root/.node-gyp/4.2.1/include/node/node_buffer.h:36:40: note: v8::MaybeLocal<v8::Object> node::Buffer::New(v8::Isolate*, char*, size_t, node::Buffer::FreeCallback, void*)
 NODE_EXTERN v8::MaybeLocal<v8::Object> New(v8::Isolate* isolate,
                                        ^
/root/.node-gyp/4.2.1/include/node/node_buffer.h:36:40: note:   candidate expects 5 arguments, 1 provided
/root/.node-gyp/4.2.1/include/node/node_buffer.h:43:40: note: v8::MaybeLocal<v8::Object> node::Buffer::New(v8::Isolate*, char*, size_t)
 NODE_EXTERN v8::MaybeLocal<v8::Object> New(v8::Isolate* isolate,
                                        ^
/root/.node-gyp/4.2.1/include/node/node_buffer.h:43:40: note:   candidate expects 3 arguments, 1 provided
i2c.target.mk:86: recipe for target 'Release/obj.target/i2c/src/i2c.o' failed
make: *** [Release/obj.target/i2c/src/i2c.o] Error 1
make: Leaving directory '/usr/local/lib/node_modules/i2c/build'
gyp ERR! build error 
gyp ERR! stack Error: `make` failed with exit code: 2
gyp ERR! stack     at ChildProcess.onExit (/usr/local/lib/node_modules/npm/node_modules/node-gyp/lib/build.js:276:23)
gyp ERR! stack     at emitTwo (events.js:87:13)
gyp ERR! stack     at ChildProcess.emit (events.js:172:7)
gyp ERR! stack     at Process.ChildProcess._handle.onexit (internal/child_process.js:200:12)
gyp ERR! System Linux 4.1.19-v7+
gyp ERR! command "/usr/local/bin/node" "/usr/local/lib/node_modules/npm/node_modules/node-gyp/bin/node-gyp.js" "rebuild"
gyp ERR! cwd /usr/local/lib/node_modules/i2c
gyp ERR! node -v v4.2.1
gyp ERR! node-gyp -v v3.3.1
gyp ERR! not ok 
npm ERR! Linux 4.1.19-v7+
npm ERR! argv "/usr/local/bin/node" "/usr/local/bin/npm" "install" "-g" "--unsafe-perm" "i2c"
npm ERR! node v4.2.1
npm ERR! npm  v3.8.6
npm ERR! code ELIFECYCLE

npm ERR! i2c@0.2.1 install: `node-gyp rebuild`
npm ERR! Exit status 1
npm ERR! 
npm ERR! Failed at the i2c@0.2.1 install script 'node-gyp rebuild'.
npm ERR! Make sure you have the latest version of node.js and npm installed.
npm ERR! If you do, this is most likely a problem with the i2c package,
npm ERR! not with npm itself.
npm ERR! Tell the author that this fails on your system:
npm ERR!     node-gyp rebuild
npm ERR! You can get information on how to open an issue for this project with:
npm ERR!     npm bugs i2c
npm ERR! Or if that isn't available, you can get their info via:
npm ERR!     npm owner ls i2c
npm ERR! There is likely additional logging output above.

npm ERR! Please include the following file with any support request:
npm ERR!     /home/pi/npm-debug.log

goog, 2016-04-13:
[npm-debug.txt](https://github.com/kelly/node-i2c/files/217068/npm-debug.txt)

infusion, 2016-04-29:
I have the same problem on node v4.2.1 and npm  v2.14.7.

doorbash, 2016-06-03:
I ran to same problem and then I gave [rasp2c](https://github.com/ni-c/node-rasp2c) a try.
worked very well.

infusion, 2016-06-04:
Thanks @doorbash, but I don't like, that it forks a new process for every setting using the cli tools. A real alternative could be [i2c-bus](https://github.com/fivdi/i2c-bus) . However, I implemented the crucial parts in C now and passing the results back to node later.

alphacharlie, 2016-06-06:
For anyone having this issue on the Raspberry Pi. It's an incompatibility between the I2C package and newer versions of Node.js. As a workaround you can install Node.js version 0.12.6 (which still works with the library) from here - http://node-arm.herokuapp.com/node_archive_armhf.deb

If the author doesn't fix this soon I'll fork it and get it running for Node >= 4.x as this  package is starting to look abandoned...

leemind, 2016-06-07:
@alphacharlie -- Please do!!!

CaptainMack, 2016-06-10:
@alphacharlie Oh Charlie, please do :-)

chris03, 2016-06-10:
A fix would be appreciated, there's quite a few projects that depend on _i2c_ and are now broken.

alphacharlie, 2016-06-12:
I forked the code off and updated it to use 'nan' for buffer creation. Will someone please try this and let me know how it works for them?  [node-i2c](https://github.com/alphacharlie/node-i2c)

(Otherwise I'll have to put some hardware together to test it myself. Probably take a few days for me to get to it...)

CaptainMack, 2016-06-12:
@alphacharlie 

I have a setup laying in the office, I'll test it tomorrow and report back

Maus34, 2016-06-12:
I installed node 4.4.5 and then the newest version of node-i2c on a Rasperry 1 B and it failed.
see:
[npm-debug.txt](https://github.com/kelly/node-i2c/files/310755/npm-debug.txt)

alphacharlie, 2016-06-12:
@Maus34 - how _exactly_ did you try to install it? (you would have to download it from the link above not from npm...) 
Try - 
'sudo apt-get update'
'sudo apt-get upgrade'
`mkdir tmp`
'cd tmp'
'git clone https://github.com/alphacharlie/node-i2c'
'npm install ./node-i2c'

I don'e have a raspberry pi 1b to test with. But the above compiles and installs successfully for me on both Node 4.x and Node 6.x on the raspberry pi 2b. (Whether it actually _works_ remains to be seen. I don't have the hardware set up to test it right now...)

Maus34, 2016-06-12:
@alphacharlie
I tried it again with exactly the commands above ( new download to tmp, whitch I dont used before)
see the last messages from the terminal
[mess.txt](https://github.com/kelly/node-i2c/files/310833/mess.txt)
and the log file
[npm-debug.txt](https://github.com/kelly/node-i2c/files/310835/npm-debug.txt)
Maybe its a probem of rights

alphacharlie, 2016-06-12:
@Maus34  - Hmmm not sure what is going on... From your debug file it is trying to install it into **~/tmp/node_modules/i2c** and _failing_.  But from the commands I gave, it should just build in place in **~/tmp/node-i2c**

Maus34, 2016-06-13:
@ Alphacharlie
seems the c++11 compiler was not foud
[typescript.txt](https://github.com/kelly/node-i2c/files/312360/typescript.txt)

alphacharlie, 2016-06-13:
@Maus34 - You might do - 
`sudo apt-get install build-essential`
and try again.

Also, The code at [https://github.com/alphacharlie/node-i2c](https://github.com/alphacharlie/node-i2c) has been updated. (I finished the conversion to the Nan.)  So anyone testing the fix will  want to re-download it.

If it works for everybody I'll do a pull request and see if @kelly will merge it.

Maus34, 2016-06-13:
@alphacharlie 
I checked with gcc --version: 4.6.3
updated it to 4.8 (for C++11), google: "raspberry c++11"
gcc --version 4.8
then it worked

alphacharlie, 2016-06-13:
Huzzah!

If you get a chance to test it, please post back with any results... :-)

alphacharlie, 2016-06-13:
**Hello All** 

**Good News - Just got done testing this package on actual I2c hardware (ADS1015 ADC) and it worked for me _without errors_!** 

Looking for more developers to confirm that it also works for them. To test you can replace the dependency in the applicable Package.json file. So this:

```
"dependencies": {
  "i2c": "^0.2.1"
},
```
becomes this:
```
  "dependencies": {
    "i2c": "https://github.com/alphacharlie/node-i2c"
  }
```

then rebuild the your node app.

_**Note - if you are an end user trying to fix a broken install, then you'll probably have to do -_
`npm install force` 
_in your app directory, then go digging for the broken dependency. (or dependencies?) until the changes are merged._

kelly, 2016-06-13:
@alphacharlie awesome! Thanks for putting this together and making the switch to Nan. Send me a PR when you have a chance. I'll look everything over and get it merged in.

Maus34, 2016-06-14:
@alphacharlie 
I start my application with `node app.js` for my node module node-HDC1000 and got the error 
 `Cannot find module 'i2c'`

alphacharlie, 2016-06-14:
@Maus34 - Can you post your package.json?

closed 2016-06-15 by kelly