kelly/node-i2c #78: Install problems

Maus34, 2016-06-15:
I installed the latest version with 
`npm install i2c`
Compiling worked well with gcc 4.8 and I got the folder /home/pi/node-i2c 
But in node I do not have the module i2c: I checked with `npm -g ls` and I have no entry in npm\package.json.

kelly, 2016-06-15:
Did you install i2c globally?

```bash
sudo npm install i2c -g
```

Maus34, 2016-06-16:
I did `sudo npm install i2c -g`
and got the folder /usr/local/lib/node_modules/i2c
in  node: > `var i2c = require('i2c');`
gives me `Error: Cannot find module i2c`
I checked again with npm -g ls and in npm\package.json. 
Manually:
`var i2c = require('/usr/local/lib/node_modules/i2c');`
gives me `undefined
`
The owner of /usr/local/lib/node_modules/i2c was nobody 
I changed it to root and then to pi, but no change

Maus34, 2016-06-18:
works now. I used the wrong folder

closed 2016-06-18 by Maus34