kelly/node-i2c #79: i2c v0.2.3 fails to install on BeagleBone running Node v0.12.13

psiphi75, 2016-06-20:
My system details are:
OS: Debian 8.5, Linux 4.1.18-ti-r56
Processor/Arch: ARM
Node version: 0.12.13
i2c version: 0.2.3 (0.2.1 and earlier versions work fine)

When I run `npm install i2c` I get the following error:
```Text
> i2c@0.2.3 install /root/node_modules/i2c
> node-gyp rebuild

make: Entering directory '/root/node_modules/i2c/build'
  CXX(target) Release/obj.target/i2c/src/i2c.o
../src/i2c.cc: In function 'void Scan(const Nan::FunctionCallbackInfo<v8::Value>&)':
../src/i2c.cc:48:49: error: call of overloaded 'New(v8::Local<v8::Primitive>)' is ambiguous
   Local<Value> err = Nan::New<Value>(Nan::Null());
                                                 ^
../src/i2c.cc:48:49: note: candidates are:
In file included from ../node_modules/nan/nan_new.h:189:0,
                 from ../node_modules/nan/nan.h:196,
                 from ../src/i2c.cc:3:
../node_modules/nan/nan_implementation_12_inl.h:389:21: note: v8::Local<T> Nan::New(v8::Handle<T>) [with T = v8::Value]
 inline v8::Local<T> New(v8::Handle<T> h) {
                     ^
../node_modules/nan/nan_implementation_12_inl.h:405:21: note: v8::Local<T> Nan::New(const Nan::Global<T>&) [with T = v8::Value]
 inline v8::Local<T> New(Global<T> const& p) {
                     ^
../src/i2c.cc: In function 'void Open(const Nan::FunctionCallbackInfo<v8::Value>&)':
../src/i2c.cc:85:49: error: call of overloaded 'New(v8::Local<v8::Primitive>)' is ambiguous
   Local<Value> err = Nan::New<Value>(Nan::Null());
                                                 ^
../src/i2c.cc:85:49: note: candidates are:
In file included from ../node_modules/nan/nan_new.h:189:0,
                 from ../node_modules/nan/nan.h:196,
                 from ../src/i2c.cc:3:
../node_modules/nan/nan_implementation_12_inl.h:389:21: note: v8::Local<T> Nan::New(v8::Handle<T>) [with T = v8::Value]
 inline v8::Local<T> New(v8::Handle<T> h) {
                     ^
../node_modules/nan/nan_implementation_12_inl.h:405:21: note: v8::Local<T> Nan::New(const Nan::Global<T>&) [with T = v8::Value]
 inline v8::Local<T> New(Global<T> const& p) {
                     ^
../src/i2c.cc: In function 'void Read(const Nan::FunctionCallbackInfo<v8::Value>&)':
../src/i2c.cc:108:49: error: call of overloaded 'New(v8::Local<v8::Primitive>)' is ambiguous
   Local<Value> err = Nan::New<Value>(Nan::Null());
                                                 ^
../src/i2c.cc:108:49: note: candidates are:
In file included from ../node_modules/nan/nan_new.h:189:0,
                 from ../node_modules/nan/nan.h:196,
                 from ../src/i2c.cc:3:
../node_modules/nan/nan_implementation_12_inl.h:389:21: note: v8::Local<T> Nan::New(v8::Handle<T>) [with T = v8::Value]
 inline v8::Local<T> New(v8::Handle<T> h) {
                     ^
../node_modules/nan/nan_implementation_12_inl.h:405:21: note: v8::Local<T> Nan::New(const Nan::Global<T>&) [with T = v8::Value]
 inline v8::Local<T> New(Global<T> const& p) {
                     ^
../src/i2c.cc: In function 'void ReadByte(const Nan::FunctionCallbackInfo<v8::Value>&)':
../src/i2c.cc:131:49: error: call of overloaded 'New(v8::Local<v8::Primitive>)' is ambiguous
   Local<Value> err = Nan::New<Value>(Nan::Null());
                                                 ^
../src/i2c.cc:131:49: note: candidates are:
In file included from ../node_modules/nan/nan_new.h:189:0,
                 from ../node_modules/nan/nan.h:196,
                 from ../src/i2c.cc:3:
../node_modules/nan/nan_implementation_12_inl.h:389:21: note: v8::Local<T> Nan::New(v8::Handle<T>) [with T = v8::Value]
 inline v8::Local<T> New(v8::Handle<T> h) {
                     ^
../node_modules/nan/nan_implementation_12_inl.h:405:21: note: v8::Local<T> Nan::New(const Nan::Global<T>&) [with T = v8::Value]
 inline v8::Local<T> New(Global<T> const& p) {
                     ^
../src/i2c.cc: In function 'void ReadBlock(const Nan::FunctionCallbackInfo<v8::Value>&)':
[... 82 lines cut ...]
gyp ERR! not ok 
npm ERR! Linux 4.1.18-ti-r56
npm ERR! argv "/usr/bin/node" "/usr/bin/npm" "install" "i2c"
npm ERR! node v0.12.13
npm ERR! npm  v2.15.0
npm ERR! code ELIFECYCLE

npm ERR! i2c@0.2.3 install: `node-gyp rebuild`
npm ERR! Exit status 1
npm ERR! 
npm ERR! Failed at the i2c@0.2.3 install script 'node-gyp rebuild'.
npm ERR! This is most likely a problem with the i2c package,
npm ERR! not with npm itself.
npm ERR! Tell the author that this fails on your system:
npm ERR!     node-gyp rebuild
npm ERR! You can get information on how to open an issue for this project with:
npm ERR!     npm bugs i2c
npm ERR! Or if that isn't available, you can get their info via:
npm ERR! 
npm ERR!     npm owner ls i2c
npm ERR! There is likely additional logging output above.

npm ERR! Please include the following file with any support request:
npm ERR!     /root/npm-debug.log
```

phixMe, 2016-07-22:
I am seeing similar issues with 12.2. Anybody out there resolve this?

kelly, 2016-07-22:
Just use 0.2.1 or earlier.

phixMe, 2016-07-22:
That worked great for me. I'm on the Edison platform. Thanks!!!

kelly, 2016-10-15:
the latest version depends on new versions of node (4.0 and later).

closed 2016-10-15 by kelly