kelly/node-i2c #80: Unable to install on Raspberry Pi 3 running Raspbian Jessie

thelonious, 2016-09-10:
It looks like I'm running node v0.10.29 and npm 1.4.21. When I run `sudo npm install i2c` I get the following:

```
npm ERR! Error: EPERM, chown '/boot/overlays/node_modules/i2c/package.json'
npm ERR!  { [Error: EPERM, chown '/boot/overlays/node_modules/i2c/package.json']
npm ERR!   errno: 50,
npm ERR!   code: 'EPERM',
npm ERR!   path: '/boot/overlays/node_modules/i2c/package.json',
npm ERR!   fstream_finish_call: 'chown',
npm ERR!   fstream_type: 'File',
npm ERR!   fstream_path: '/boot/overlays/node_modules/i2c/package.json',
npm ERR!   fstream_class: 'FileWriter',
npm ERR!   fstream_stack: 
npm ERR!    [ '/usr/lib/nodejs/fstream/lib/writer.js:305:19',
npm ERR!      '/usr/lib/nodejs/graceful-fs/polyfills.js:143:7',
npm ERR!      'Object.oncomplete (evalmachine.<anonymous>:107:15)' ] }
npm ERR! 
npm ERR! Please try running this command again as root/Administrator.

npm ERR! System Linux 4.4.13-v7+
npm ERR! command "/usr/bin/nodejs" "/usr/bin/npm" "install" "i2c"
npm ERR! cwd /boot/overlays
npm ERR! node -v v0.10.29
npm ERR! npm -v 1.4.21
npm ERR! path /boot/overlays/node_modules/i2c/package.json
npm ERR! fstream_path /boot/overlays/node_modules/i2c/package.json
npm ERR! fstream_type File
npm ERR! fstream_class FileWriter
npm ERR! fstream_finish_call chown
npm ERR! code EPERM
npm ERR! errno 50
npm ERR! stack Error: EPERM, chown '/boot/overlays/node_modules/i2c/package.json'
npm ERR! fstream_stack /usr/lib/nodejs/fstream/lib/writer.js:305:19
npm ERR! fstream_stack /usr/lib/nodejs/graceful-fs/polyfills.js:143:7
npm ERR! fstream_stack Object.oncomplete (evalmachine.<anonymous>:107:15)
npm ERR! 
npm ERR! Additional logging details can be found in:
npm ERR!     /boot/overlays/npm-debug.log
npm ERR! not ok code 0
```

I see that `chown` is failing. I could probably hack permissions on the directory structure there, but that doesn't seem like a good idea. Any suggestions on how to get around this?

Thanks!

thelonious, 2016-09-10:
I got a little closer. I set a password on root and was able to `su root` and the build/install got further. I had to install coffee-script and nan, but now it's failing on a symlink step with coffeescript

```
npm ERR! Error: EPERM, symlink '../coffee-script/bin/coffee'
npm ERR!  { [Error: EPERM, symlink '../coffee-script/bin/coffee'] errno: 50, code: 'EPERM', path: '../coffee-script/bin/coffee' }
npm ERR! 
npm ERR! Please try running this command again as root/Administrator.

npm ERR! System Linux 4.4.13-v7+
npm ERR! command "/usr/bin/nodejs" "/usr/bin/npm" "install" "i2c"
npm ERR! cwd /boot/overlays
npm ERR! node -v v0.10.29
npm ERR! npm -v 1.4.21
npm ERR! path ../coffee-script/bin/coffee
npm ERR! code EPERM
npm ERR! errno 50
npm ERR! stack Error: EPERM, symlink '../coffee-script/bin/coffee'

npm ERR! 
npm ERR! Additional logging details can be found in:
npm ERR!     /boot/overlays/npm-debug.log
npm ERR! not ok code 0
```

thelonious, 2016-09-10:
I tried a clean install of Raspian. I have the same versions of node and npm. Here's the failure I see now:

```
gyp ERR! build error 
gyp ERR! stack Error: `make` failed with exit code: 2
gyp ERR! stack     at ChildProcess.onExit (/usr/share/node-gyp/lib/build.js:267:23)
gyp ERR! stack     at ChildProcess.emit (events.js:98:17)
gyp ERR! stack     at Process.ChildProcess._handle.onexit (child_process.js:809:12)
gyp ERR! System Linux 4.4.13-v7+
gyp ERR! command "nodejs" "/usr/bin/node-gyp" "rebuild"
gyp ERR! cwd /home/pi/Documents/node/node_modules/i2c
gyp ERR! node -v v0.10.29
gyp ERR! node-gyp -v v0.12.2
gyp ERR! not ok 
npm WARN This failure might be due to the use of legacy binary "node"
npm WARN For further explanations, please read
/usr/share/doc/nodejs/README.Debian
 
npm ERR! i2c@0.2.3 install: `node-gyp rebuild`
npm ERR! Exit status 1
npm ERR! 
npm ERR! Failed at the i2c@0.2.3 install script.
npm ERR! This is most likely a problem with the i2c package,
npm ERR! not with npm itself.
npm ERR! Tell the author that this fails on your system:
npm ERR!     node-gyp rebuild
npm ERR! You can get their info via:
npm ERR!     npm owner ls i2c
npm ERR! There is likely additional logging output above.

npm ERR! System Linux 4.4.13-v7+
npm ERR! command "/usr/bin/nodejs" "/usr/bin/npm" "install" "i2c"
npm ERR! cwd /home/pi/Documents/node
npm ERR! node -v v0.10.29
npm ERR! npm -v 1.4.21
npm ERR! code ELIFECYCLE
npm ERR! 
npm ERR! Additional logging details can be found in:
npm ERR!     /home/pi/Documents/node/npm-debug.log
npm ERR! not ok code 0
```

nielsnl68, 2016-10-08:
As the readme says, this version does not support the node.js version you use. Upgrade node.js to 4.6 or use the 0.18 version of i2c.

thelonious, 2016-10-08:
Ah, yes. I didn't read down that far. It's too bad the default experience on Raspian isn't nicer for node. I've switched over to Python for now. Thanks!

closed 2016-10-08 by thelonious

nielsnl68, 2016-10-08:
Why not upgrading nodejs to 4.6 or above?

thelonious, 2016-10-10:
I ran out of time with that project and needed a solution right then. I may give node a try next time around. FWIW, I have 'n' running now, so it's easy switch versions. I was able to build i2c with 5.12.0 and 6.6.0 just fine on the Pi 3.