kelly/node-i2c #83: node-red-contrib-bmp085

liborp, 2017-04-02:
During the install o node-red-contrib-bmp085 module I had a strange error. Have you any advice where the problem is?
Thank you in advance.
LP.

2 Apr 10:08:53 - [info] Installing module: node-red-contrib-bmp085
2 Apr 10:09:37 - [warn] Installation of module node-red-contrib-bmp085 failed:
2 Apr 10:09:37 - [warn] ------------------------------------------
2 Apr 10:09:37 - [warn] Error: Command failed: npm install --production node-red-contrib-bmp085
../src/i2c.cc: In function ‘void setAddress(int8_t)’:
../src/i2c.cc:21:28: error: ‘New’ is not a member of ‘v8::String’
Exception::TypeError(String::New("Failed to set address"))
^
../src/i2c.cc:22:5: error: ‘ThrowException’ was not declared in this scope
);
^
../src/i2c.cc: At global scope:
../src/i2c.cc:26:32: error: ‘Arguments’ does not name a type
Handle<Value> SetAddress(const Arguments& args) {
^
In file included from /home/pi/.node-gyp/7.7.1/include/node/node.h:42:0,
from ../src/i2c.cc:1:
/home/pi/.node-gyp/7.7.1/include/node/v8.h: In function ‘v8::Handle<v8::Value> SetAddress(const int&)’:
/home/pi/.node-gyp/7.7.1/include/node/v8.h:908:13: error: ‘v8::HandleScope::HandleScope()’ is protected
V8_INLINE HandleScope() {}
^
../src/i2c.cc:27:15: error: within this context
HandleScope scope;
^
../src/i2c.cc:29:16: error: invalid types ‘const int[int]’ for array subscript
addr = args[0]->Int32Value();
^
../src/i2c.cc:32:16: error: ‘class v8::HandleScope’ has no member named ‘Close’
return scope.Close(Undefined());
^
../src/i2c.cc:32:32: error: too few arguments to function ‘v8::Local<v8::Primitive> v8::Undefined(v8::Isolate*)’
return scope.Close(Undefined());
^
In file included from /home/pi/.node-gyp/7.7.1/include/node/node.h:42:0,
from ../src/i2c.cc:1:
/home/pi/.node-gyp/7.7.1/include/node/v8.h:327:27: note: declared here
friend Local<Primitive> Undefined(Isolate* isolate);
^
../src/i2c.cc: At global scope:
../src/i2c.cc:35:26: error: ‘Arguments’ does not name a type
Handle<Value> Scan(const Arguments& args) {
^
In file included from /home/pi/.node-gyp/7.7.1/include/node/node.h:42:0,
from ../src/i2c.cc:1:
/home/pi/.node-gyp/7.7.1/include/node/v8.h: In function ‘v8::Handle<v8::Value> Scan(const int&)’:
/home/pi/.node-gyp/7.7.1/include/node/v8.h:908:13: error: ‘v8::HandleScope::HandleScope()’ is protected
V8_INLINE HandleScope() {}
^
../src/i2c.cc:36:15: error: within this context
HandleScope scope;
^
../src/i2c.cc:39:58: error: invalid types ‘const int[int]’ for array subscript
Local<Function> callback = Local<Function>::Cast(args[0]);
^
../src/i2c.cc:40:38: error: invalid conversion from ‘int’ to ‘v8::Isolate*’ [-fpermissive]
Local<Array> results(Array::New(128));
^
In file included from /home/pi/.node-gyp/7.7.1/include/node/node.h:42:0,
from ../src/i2c.cc:1:
/home/pi/.node-gyp/7.7.1/include/node/v8.h:3354:23: note: initializing argument 1 of ‘static v8::Local<v8::Array> v8::Array::New(v8::Isolate*, int)’
static Local<Array> New(Isolate* isolate, int length = 0);
^
../src/i2c.cc:41:45: error: too few arguments to function ‘v8::Local<v8::Primitive> v8::Null(v8::Isolate*)’
Local<Value> err = Local<Value>::New(Null());
^
In file included from /home/pi/.node-gyp/7.7.1/include/node/node.h:42:0,
[... 661 lines cut ...]
../src/i2c.cc: In function ‘v8::Handle<v8::Value> Write(const int&)’:
../src/i2c.cc:212:1: warning: control reaches end of non-void function [-Wreturn-type]
}
^
../src/i2c.cc: In function ‘v8::Handle<v8::Value> WriteByte(const int&)’:
../src/i2c.cc:233:1: warning: control reaches end of non-void function [-Wreturn-type]
}
^
../src/i2c.cc: In function ‘v8::Handle<v8::Value> WriteBlock(const int&)’:
../src/i2c.cc:259:1: warning: control reaches end of non-void function [-Wreturn-type]
}
^
../src/i2c.cc: In function ‘v8::Handle<v8::Value> WriteWord(const int&)’:
../src/i2c.cc:282:1: warning: control reaches end of non-void function [-Wreturn-type]
}
^
make: *** [Release/obj.target/i2c/src/i2c.o] Error 1
gyp ERR! build error
gyp ERR! stack Error: `make` failed with exit code: 2
gyp ERR! stack     at ChildProcess.onExit (/opt/nodejs/lib/node_modules/npm/node_modules/node-gyp/lib/build.js:276:23)
gyp ERR! stack     at emitTwo (events.js:106:13)
gyp ERR! stack     at ChildProcess.emit (events.js:194:7)
gyp ERR! stack     at Process.ChildProcess._handle.onexit (internal/child_process.js:215:12)
gyp ERR! System Linux 4.4.50+
gyp ERR! command "/opt/nodejs/bin/node" "/opt/nodejs/lib/node_modules/npm/node_modules/node-gyp/bin/node-gyp.js" "rebuild"

kelly, 2017-04-04:
This package is most likely depending on an older version of i2c that isn't compatible with the newer node version.

closed 2017-04-04 by kelly

liborp, 2017-04-04:
Thank you Kelly for the information.

All the best

Libor

kelly, 2017-04-04:
If you fork the package, you likely can update i2c to a newer version and it should work correctly.