kelly/node-i2c #84: npm install i2c fails on BeagleBone

billtrudell, 2017-04-07:
An install of i2c on a Beaglebone Green, with relatively new/updated Nodejs and NPM, is failing?  Any clues?

verbose stack Error: i2c@0.2.3 install: `node-gyp rebuild`
verbose stack spawn ENOENT
verbose stack     at ChildProcess.<anonymous> (/root/nodejs/lib/node_modules/npm/lib/utils/spawn.js:33:16)
verbose stack     at emitTwo (events.js:106:13)
verbose stack     at ChildProcess.emit (events.js:194:7)
verbose stack     at maybeClose (internal/child_process.js:899:16)
verbose stack     at Process.ChildProcess._handle.onexit (internal/child_process.js:226:5)
verbose pkgid i2c@0.2.3
verbose cwd /home/examples.i2c
error Linux 4.4.9-ti-r25
error argv "/root/nodejs/bin/node" "/usr/local/bin/npm" "install" "i2c"
error node v7.8.0
error npm  v4.2.0
error file sh
error code ELIFECYCLE
error errno ENOENT
error syscall spawn
error i2c@0.2.3 install: `node-gyp rebuild`
error spawn ENOENT
error Failed at the i2c@0.2.3 install script 'node-gyp rebuild'.
[2017-04-07T19_39_40_305Z-debug.txt](https://github.com/kelly/node-i2c/files/906731/2017-04-07T19_39_40_305Z-debug.txt)

joeboyscout04, 2017-09-11:
I'm getting similar errors with "npm install i2c" on both my mac and raspberry pi:

`npm ERR! i2c@0.2.3 install: `node-gyp rebuild`
npm ERR! Exit status 1
npm ERR! 
npm ERR! Failed at the i2c@0.2.3 install script.
npm ERR! This is most likely a problem with the i2c package,
npm ERR! not with npm itself.
npm ERR! Tell the author that this fails on your system:
npm ERR!     node-gyp rebuild
npm ERR! You can get their info via:
npm ERR!     npm owner ls i2c
npm ERR! There is likely additional logging output above.

npm ERR! System Linux 4.9.41+
npm ERR! command "/usr/bin/nodejs" "/usr/bin/npm" "install" "piglow"
npm ERR! cwd /home/pi
npm ERR! node -v v4.8.2
npm ERR! npm -v 1.4.21
npm ERR! code ELIFECYCLE
npm ERR! 
npm ERR! Additional logging details can be found in:
npm ERR!     /home/pi/npm-debug.log`

OGVCelsius, 2018-07-20:
Did you solve the problem? Because I have the same happening right now. My node version is 8.11.2 and my     "body-parser": "^1.18.3",
    "express": "^4.16.3",
    "mongodb": "^3.1.1",
    "mqtt": "^2.18.3",
    "npm": "^6.2.0",
    "server": "^1.0.18"