kelly/node-i2c #87: Unable to install in orange pi zero

dedhdude, 2017-05-17:
root@orangepizero:~# sudo npm install i2c-bus
npm ERR! Linux 3.4.113-sun8i
npm ERR! argv "/usr/bin/nodejs" "/usr/bin/npm" "install" "i2c-bus"
npm ERR! node v6.10.3
npm ERR! npm  v3.10.10
npm ERR! file /root/.npm/bindings/1.2.1/package/package.json
npm ERR! code EJSONPARSE

npm ERR! Failed to parse json
npm ERR! Unexpected end of input at 49:59
npm ERR!     "url": "https://github.com/TooTallNate/node-bindings/i
npm ERR!                                                           ^
npm ERR! File: /root/.npm/bindings/1.2.1/package/package.json
npm ERR! Failed to parse package.json data.
npm ERR! package.json must be actual JSON, not just JavaScript.
npm ERR!
npm ERR! This is not a bug in npm.
npm ERR! Tell the package author to fix their package.json file. JSON.parse

npm ERR! Please include the following file with any support request:
npm ERR!     /root/npm-debug.log

kelly, 2017-05-17:
wrong module

closed 2017-05-17 by kelly

dedhdude, 2017-05-17:
That I know but how to solve the problem.
How should I rebuild....?