kelly/node-i2c #90: Not working with NodeJS v10.4.0

thorleifjaocbsen, 2018-06-10:
```npm WARN deprecated coffee-script@1.9.1: CoffeeScript on NPM has moved to "coffeescript" (no hyphen)

> i2c@0.2.3 install /home/pi/NodeROV/node_modules/i2c
> node-gyp rebuild

make: Entering directory '/home/pi/NodeROV/node_modules/i2c/build'
  CXX(target) Release/obj.target/i2c/src/i2c.o
In file included from ../../nan/nan.h:190:0,
                 from ../src/i2c.cc:3:
../../nan/nan_maybe_43_inl.h: In function 'Nan::Maybe<bool> Nan::ForceSet(v8::Local<v8::Object>, v8::Local<v8::Value>, v8::Local<v8::Value>, v8::PropertyAttribute)':
../../nan/nan_maybe_43_inl.h:88:15: error: 'class v8::Object' has no member named 'ForceSet'
   return obj->ForceSet(GetCurrentContext(), key, value, attribs);
               ^~~~~~~~
In file included from ../src/i2c.cc:3:0:
../../nan/nan.h: In function 'v8::Local<v8::Value> Nan::MakeCallback(v8::Local<v8::Object>, v8::Local<v8::Function>, int, v8::Local<v8::Value>*)':
../../nan/nan.h:817:60: warning: 'v8::Local<v8::Value> node::MakeCallback(v8::Isolate*, v8::Local<v8::Object>, v8::Local<v8::Function>, int, v8::Local<v8::Value>*)' is deprecated: Use MakeCallback(..., async_context) [-Wdeprecated-declarations]
         v8::Isolate::GetCurrent(), target, func, argc, argv);
                                                            ^
In file included from ../src/i2c.cc:1:0:
/home/pi/.node-gyp/10.4.0/include/node/node.h:172:50: note: declared here
                 NODE_EXTERN v8::Local<v8::Value> MakeCallback(
                                                  ^
/home/pi/.node-gyp/10.4.0/include/node/node.h:88:42: note: in definition of macro 'NODE_DEPRECATED'
     __attribute__((deprecated(message))) declarator
                                          ^~~~~~~~~~
In file included from ../src/i2c.cc:3:0:
../../nan/nan.h: In function 'v8::Local<v8::Value> Nan::MakeCallback(v8::Local<v8::Object>, v8::Local<v8::String>, int, v8::Local<v8::Value>*)':
../../nan/nan.h:831:62: warning: 'v8::Local<v8::Value> node::MakeCallback(v8::Isolate*, v8::Local<v8::Object>, v8::Local<v8::String>, int, v8::Local<v8::Value>*)' is deprecated: Use MakeCallback(..., async_context) [-Wdeprecated-declarations]
         v8::Isolate::GetCurrent(), target, symbol, argc, argv);
                                                              ^
In file included from ../src/i2c.cc:1:0:
/home/pi/.node-gyp/10.4.0/include/node/node.h:165:50: note: declared here
                 NODE_EXTERN v8::Local<v8::Value> MakeCallback(
                                                  ^
/home/pi/.node-gyp/10.4.0/include/node/node.h:88:42: note: in definition of macro 'NODE_DEPRECATED'
     __attribute__((deprecated(message))) declarator
                                          ^~~~~~~~~~
In file included from ../src/i2c.cc:3:0:
../../nan/nan.h: In function 'v8::Local<v8::Value> Nan::MakeCallback(v8::Local<v8::Object>, const char*, int, v8::Local<v8::Value>*)':
../../nan/nan.h:845:62: warning: 'v8::Local<v8::Value> node::MakeCallback(v8::Isolate*, v8::Local<v8::Object>, const char*, int, v8::Local<v8::Value>*)' is deprecated: Use MakeCallback(..., async_context) [-Wdeprecated-declarations]
         v8::Isolate::GetCurrent(), target, method, argc, argv);
                                                              ^
In file included from ../src/i2c.cc:1:0:
/home/pi/.node-gyp/10.4.0/include/node/node.h:158:50: note: declared here
                 NODE_EXTERN v8::Local<v8::Value> MakeCallback(
                                                  ^
/home/pi/.node-gyp/10.4.0/include/node/node.h:88:42: note: in definition of macro 'NODE_DEPRECATED'
     __attribute__((deprecated(message))) declarator
                                          ^~~~~~~~~~
In file included from ../src/i2c.cc:3:0:
../../nan/nan.h: In member function 'v8::Local<v8::Value> Nan::Callback::Call_(v8::Isolate*, v8::Local<v8::Object>, int, v8::Local<v8::Value>*) const':
../../nan/nan.h:1463:5: warning: 'v8::Local<v8::Value> node::MakeCallback(v8::Isolate*, v8::Local<v8::Object>, v8::Local<v8::Function>, int, v8::Local<v8::Value>*)' is deprecated: Use MakeCallback(..., async_context) [-Wdeprecated-declarations]
     ));
     ^
In file included from ../src/i2c.cc:1:0:
/home/pi/.node-gyp/10.4.0/include/node/node.h:172:50: note: declared here
                 NODE_EXTERN v8::Local<v8::Value> MakeCallback(
                                                  ^
/home/pi/.node-gyp/10.4.0/include/node/node.h:88:42: note: in definition of macro 'NODE_DEPRECATED'
     __attribute__((deprecated(message))) declarator
                                          ^~~~~~~~~~
../src/i2c.cc: In function 'void Open(const Nan::FunctionCallbackInfo<v8::Value>&)':
../src/i2c.cc:84:35: warning: 'v8::String::Utf8Value::Utf8Value(v8::Local<v8::Value>)' is deprecated: Use Isolate version [-Wdeprecated-declarations]
   String::Utf8Value device(info[0]);
                                   ^
In file included from /home/pi/.node-gyp/10.4.0/include/node/v8.h:26:0,
                 from /home/pi/.node-gyp/10.4.0/include/node/node.h:63,
                 from ../src/i2c.cc:1:
/home/pi/.node-gyp/10.4.0/include/node/v8.h:2846:28: note: declared here
                   explicit Utf8Value(Local<v8::Value> obj));
                            ^
/home/pi/.node-gyp/10.4.0/include/node/v8config.h:324:3: note: in definition of macro 'V8_DEPRECATED'
   declarator __attribute__((deprecated(message)))
   ^~~~~~~~~~
i2c.target.mk:93: recipe for target 'Release/obj.target/i2c/src/i2c.o' failed
make: *** [Release/obj.target/i2c/src/i2c.o] Error 1
make: Leaving directory '/home/pi/NodeROV/node_modules/i2c/build'
gyp ERR! build error 
gyp ERR! stack Error: `make` failed with exit code: 2
gyp ERR! stack     at ChildProcess.onExit (/opt/nodejs/lib/node_modules/npm/node_modules/node-gyp/lib/build.js:258:23)
gyp ERR! stack     at ChildProcess.emit (events.js:182:13)
gyp ERR! stack     at Process.ChildProcess._handle.onexit (internal/child_process.js:237:12)
gyp ERR! System Linux 4.14.34-v7+
gyp ERR! command "/opt/nodejs/bin/node" "/opt/nodejs/lib/node_modules/npm/node_modules/node-gyp/bin/node-gyp.js" "rebuild"
gyp ERR! cwd /home/pi/NodeROV/node_modules/i2c
gyp ERR! node -v v10.4.0
gyp ERR! node-gyp -v v3.6.2
gyp ERR! not ok 
npm WARN noderov@1.0.0 No repository field.

npm ERR! code ELIFECYCLE
npm ERR! errno 1
npm ERR! i2c@0.2.3 install: `node-gyp rebuild`
npm ERR! Exit status 1
npm ERR! 
npm ERR! Failed at the i2c@0.2.3 install script.
npm ERR! This is probably not a problem with npm. There is likely additional logging output above.

npm ERR! A complete log of this run can be found in:
npm ERR!     /home/pi/.npm/_logs/2018-06-10T12_39_35_129Z-debug.log
```

rzr, 2018-08-08:
As (wrongly) reported elsewhere:
https://github.com/fivdi/i2c-bus/issues/52

I confirm i2c module failed in node:latest (10) docker image while it's ok on node:9.

rzr, 2018-08-21:
Latest known working version is v9.11.2 (v10.0.0 and v10.9.0 failed using nvm)

Is anyone active on this issue ?
or do you know other i2c implementations working for node10 ?

stheine, 2018-09-06:
I have been switching to https://www.npmjs.com/package/rpio . this is working with i2c on node 10.

matteliot, 2018-09-23:
Node 9.11.2 works for me- 10.x hasn't worked at all for it. Latest NPM, etc, npm rebuild i2c, nothing worked.

rzr, 2018-11-09:
Full log of failure at:
https://travis-ci.org/rzr/generic-sensors-lite/jobs/452971481

Is anyone working on this ?

rzr, 2018-11-23:
It looks like this package is more active than i2c:

* https://www.npmjs.com/package/i2c-bus
* https://www.npmjs.com/package/i2c

Is there a way to make "i2c" be cooperatively maintained  ? @kelly ?

matteliot, 2018-11-23:
I swapped to i2c-bus it wasn't too big an issue.

rzr, 2018-11-23:
Do you think both could merge to make a smooth transition ?

matteliot, 2018-11-23:
doubt it, it'd be better to select one and do into the future with it, I think

kelly, 2018-11-23:
Feel free to submit a PR to fix issues. I'm on vacation right now, an will
look over this thread when I return.

rzr, 2018-12-20:
Is anyone interested into co maintenance in this team :

https://www.npmjs.com/org/abandonware

Silvia-t, 2019-01-29:
Also having trouble here on a RPi with Raspbian Stretch 9, NodeJS v10.15.0 and npm 6.4.1..

```
pi@raspberrypi:~ $ sudo npm install i2c
npm WARN deprecated coffee-script@1.9.1: CoffeeScript on NPM has moved to "coffeescript" (no hyphen)

> i2c@0.2.3 install /home/pi/node_modules/i2c
> node-gyp rebuild

gyp WARN EACCES user "root" does not have permission to access the dev dir "/root/.node-gyp/10.15.0"
gyp WARN EACCES attempting to reinstall using temporary dev dir "/home/pi/node_modules/i2c/.node-gyp"
gyp WARN install got an error, rolling back install
gyp WARN install got an error, rolling back install
gyp ERR! configure error
gyp ERR! stack Error: EACCES: permission denied, mkdir '/home/pi/node_modules/i2c/.node-gyp'
gyp ERR! System Linux 4.14.79-v7+
gyp ERR! command "/usr/bin/node" "/usr/lib/node_modules/npm/node_modules/node-gyp/bin/node-gyp.js" "rebuild"
gyp ERR! cwd /home/pi/node_modules/i2c
gyp ERR! node -v v10.15.0
gyp ERR! node-gyp -v v3.8.0
gyp ERR! not ok
npm WARN enoent ENOENT: no such file or directory, open '/home/pi/package.json'
npm WARN pi No description
npm WARN pi No repository field.
npm WARN pi No README data
npm WARN pi No license field.

npm ERR! code ELIFECYCLE
npm ERR! errno 1
npm ERR! i2c@0.2.3 install: `node-gyp rebuild`
npm ERR! Exit status 1
npm ERR!
npm ERR! Failed at the i2c@0.2.3 install script.
npm ERR! This is probably not a problem with npm. There is likely additional log

npm ERR! A complete log of this run can be found in:
npm ERR!     /root/.npm/_logs/2019-01-29T15_54_10_558Z-debug.log

```

Log:

```
(0 info it worked if it ends with ok
1 verbose cli [ '/usr/bin/node', '/usr/bin/npm', 'install', 'i2c' ]
2 info using npm@6.4.1
3 info using node@v10.15.0
4 verbose npm-session 7029f1dc5b5105c6
5 silly install loadCurrentTree
6 silly install readLocalPackageData
7 http fetch GET 304 https://registry.npmjs.org/i2c 452ms (from cache)
8 silly pacote tag manifest for i2c@latest fetched in 519ms
9 timing stage:loadCurrentTree Completed in 1307ms
10 silly install loadIdealTree
11 silly install cloneCurrentTreeToIdealTree
12 timing stage:loadIdealTree:cloneCurrentTree Completed in 2ms
13 silly install loadShrinkwrap
14 timing stage:loadIdealTree:loadShrinkwrap Completed in 3ms
15 silly install loadAllDepsIntoIdealTree
16 silly resolveWithNewModule i2c@0.2.3 checking installable status
17 http fetch GET 304 https://registry.npmjs.org/bindings 172ms (from cache)
18 silly pacote version manifest for bindings@1.2.1 fetched in 182ms
19 silly resolveWithNewModule bindings@1.2.1 checking installable status
20 http fetch GET 304 https://registry.npmjs.org/nan 269ms (from cache)
21 silly pacote range manifest for nan@~2.3.5 fetched in 293ms
22 silly resolveWithNewModule nan@2.3.5 checking installable status
23 http fetch GET 304 https://registry.npmjs.org/repl 301ms (from cache)
24 http fetch GET 304 https://registry.npmjs.org/coffee-script 309ms (from cache)
25 silly pacote version manifest for repl@0.1.3 fetched in 315ms
26 silly resolveWithNewModule repl@0.1.3 checking installable status
27 silly pacote version manifest for coffee-script@1.9.1 fetched in 326ms
28 warn deprecated coffee-script@1.9.1: CoffeeScript on NPM has moved to "coffeescript" (no hyphen)
29 silly resolveWithNewModule coffee-script@1.9.1 checking installable status
30 http fetch GET 304 https://registry.npmjs.org/underscore 336ms (from cache)
31 silly pacote version manifest for underscore@1.8.2 fetched in 344ms
32 silly resolveWithNewModule underscore@1.8.2 checking installable status
33 timing stage:loadIdealTree:loadAllDepsIntoIdealTree Completed in 406ms
34 timing stage:loadIdealTree Completed in 430ms
35 silly currentTree pi
36 silly idealTree pi
36 silly idealTree ├── bindings@1.2.1
36 silly idealTree ├── coffee-script@1.9.1
36 silly idealTree ├── i2c@0.2.3
36 silly idealTree ├── nan@2.3.5
36 silly idealTree ├── repl@0.1.3
36 silly idealTree └── underscore@1.8.2
37 silly install generateActionsToTake
38 timing stage:generateActionsToTake Completed in 27ms
39 silly diffTrees action count 6
40 silly diffTrees add bindings@1.2.1
41 silly diffTrees add coffee-script@1.9.1
42 silly diffTrees add nan@2.3.5
43 silly diffTrees add repl@0.1.3
44 silly diffTrees add underscore@1.8.2
45 silly diffTrees add i2c@0.2.3
46 silly decomposeActions action count 48
47 silly decomposeActions fetch bindings@1.2.1
48 silly decomposeActions extract bindings@1.2.1
49 silly decomposeActions preinstall bindings@1.2.1
50 silly decomposeActions build bindings@1.2.1
51 silly decomposeActions install bindings@1.2.1
```

Thanks :)

Silvia-t, 2019-01-30:
Hi again,

I've sorted out the permissions issue and now I'm getting the same problems as @thorleifjaocbsen:

```
pi@raspberrypi:~/node_modules $ npm install i2c
npm WARN deprecated coffee-script@1.9.1: CoffeeScript on NPM has moved to "coffeescript" (no hyphen)

> i2c@0.2.3 install /home/pi/node_modules/i2c
> node-gyp rebuild

make: Entering directory '/home/pi/node_modules/i2c/build'
  CXX(target) Release/obj.target/i2c/src/i2c.o
In file included from ../node_modules/nan/nan.h:190:0,
                 from ../src/i2c.cc:3:
../node_modules/nan/nan_maybe_43_inl.h: In function ‘Nan::Maybe<bool> Nan::ForceSet(v8::Local<v8::Object>, v8::Local<v8::Value>, v8::Local<v8::Value>, v8::PropertyAttribute)’:
../node_modules/nan/nan_maybe_43_inl.h:88:15: error: ‘class v8::Object’ has no member named ‘ForceSet’
   return obj->ForceSet(GetCurrentContext(), key, value, attribs);
               ^~~~~~~~
In file included from ../src/i2c.cc:3:0:
../node_modules/nan/nan.h: In function ‘v8::Local<v8::Value> Nan::MakeCallback(v8::Local<v8::Object>, v8::Local<v8::Function>, int, v8::Local<v8::Value>*)’:
../node_modules/nan/nan.h:817:60: warning: ‘v8::Local<v8::Value> node::MakeCallback(v8::Isolate*, v8::Local<v8::Object>, v8::Local<v8::Function>, int, v8::Local<v8::Value>*)’ is deprecated: Use MakeCallback(..., async_context) [-Wdeprecated-declarations]
         v8::Isolate::GetCurrent(), target, func, argc, argv);
                                                            ^
In file included from ../src/i2c.cc:1:0:
/home/pi/.node-gyp/10.15.0/include/node/node.h:177:50: note: declared here
                 NODE_EXTERN v8::Local<v8::Value> MakeCallback(
                                                  ^
/home/pi/.node-gyp/10.15.0/include/node/node.h:91:42: note: in definition of macro ‘NODE_DEPRECATED’
     __attribute__((deprecated(message))) declarator
                                          ^~~~~~~~~~
In file included from ../src/i2c.cc:3:0:
../node_modules/nan/nan.h: In function ‘v8::Local<v8::Value> Nan::MakeCallback(v8::Local<v8::Object>, v8::Local<v8::String>, int, v8::Local<v8::Value>*)’:
../node_modules/nan/nan.h:831:62: warning: ‘v8::Local<v8::Value> node::MakeCallback(v8::Isolate*, v8::Local<v8::Object>, v8::Local<v8::String>, int, v8::Local<v8::Value>*)’ is deprecated: Use MakeCallback(..., async_context) [-Wdeprecated-declarations]
         v8::Isolate::GetCurrent(), target, symbol, argc, argv);
                                                              ^
In file included from ../src/i2c.cc:1:0:
/home/pi/.node-gyp/10.15.0/include/node/node.h:170:50: note: declared here
                 NODE_EXTERN v8::Local<v8::Value> MakeCallback(
                                                  ^
/home/pi/.node-gyp/10.15.0/include/node/node.h:91:42: note: in definition of macro ‘NODE_DEPRECATED’
     __attribute__((deprecated(message))) declarator
                                          ^~~~~~~~~~
In file included from ../src/i2c.cc:3:0:
../node_modules/nan/nan.h: In function ‘v8::Local<v8::Value> Nan::MakeCallback(v8::Local<v8::Object>, const char*, int, v8::Local<v8::Value>*)’:
../node_modules/nan/nan.h:845:62: warning: ‘v8::Local<v8::Value> node::MakeCallback(v8::Isolate*, v8::Local<v8::Object>, const char*, int, v8::Local<v8::Value>*)’ is deprecated: Use MakeCallback(..., async_context) [-Wdeprecated-declarations]
         v8::Isolate::GetCurrent(), target, method, argc, argv);
                                                              ^
In file included from ../src/i2c.cc:1:0:
/home/pi/.node-gyp/10.15.0/include/node/node.h:163:50: note: declared here
                 NODE_EXTERN v8::Local<v8::Value> MakeCallback(
                                                  ^
/home/pi/.node-gyp/10.15.0/include/node/node.h:91:42: note: in definition of macro ‘NODE_DEPRECATED’
     __attribute__((deprecated(message))) declarator
                                          ^~~~~~~~~~
In file included from ../src/i2c.cc:3:0:
../node_modules/nan/nan.h: In member function ‘v8::Local<v8::Value> Nan::Callback::Call_(v8::Isolate*, v8::Local<v8::Object>, int, v8::Local<v8::Value>*) const’:
../node_modules/nan/nan.h:1463:5: warning: ‘v8::Local<v8::Value> node::MakeCallback(v8::Isolate*, v8::Local<v8::Object>, v8::Local<v8::Function>, int, v8::Local<v8::Value>*)’ is deprecated: Use MakeCallback(..., async_context) [-Wdeprecated-declarations]
     ));
     ^
In file included from ../src/i2c.cc:1:0:
/home/pi/.node-gyp/10.15.0/include/node/node.h:177:50: note: declared here
                 NODE_EXTERN v8::Local<v8::Value> MakeCallback(
                                                  ^
/home/pi/.node-gyp/10.15.0/include/node/node.h:91:42: note: in definition of macro ‘NODE_DEPRECATED’
     __attribute__((deprecated(message))) declarator
                                          ^~~~~~~~~~
../src/i2c.cc: In function ‘void SetAddress(const Nan::FunctionCallbackInfo<v8::Value>&)’:
../src/i2c.cc:38:30: warning: ‘int32_t v8::Value::Int32Value() const’ is deprecated: Use maybe version [-Wdeprecated-declarations]
   addr = info[0]->Int32Value();
[... 166 lines cut ...]
gyp ERR! command "/usr/bin/node" "/usr/lib/node_modules/npm/node_modules/node-gyp/bin/node-gyp.js" "rebuild"
gyp ERR! cwd /home/pi/node_modules/i2c
gyp ERR! node -v v10.15.0
gyp ERR! node-gyp -v v3.8.0
gyp ERR! not ok
npm WARN enoent ENOENT: no such file or directory, open '/home/pi/package.json'
npm WARN pi No description
npm WARN pi No repository field.
npm WARN pi No README data
npm WARN pi No license field.

npm ERR! code ELIFECYCLE
npm ERR! errno 1
npm ERR! i2c@0.2.3 install: `node-gyp rebuild`
npm ERR! Exit status 1
npm ERR!
npm ERR! Failed at the i2c@0.2.3 install script.
npm ERR! This is probably not a problem with npm. There is likely additional logging output above.

npm ERR! A complete log of this run can be found in:
npm ERR!     /home/pi/.npm/_logs/2019-01-30T09_16_27_268Z-debug.log

```

Thanks.

rzr, 2019-02-25:
As anyone fixed this in a fork ?

jscaglia, 2019-09-25:
Hello, I tried to install this library but does not working in Node 10.15.2
I need this library because is ones of PN532 dependencies.
The error that I have is ...
Do you know if exist any solution??

make: se entra en el directorio '/home/pi/noderest/cedula/node_modules/i2c/build'
  CXX(target) Release/obj.target/i2c/src/i2c.o
In file included from ../node_modules/nan/nan.h:190,
                 from ../src/i2c.cc:3:
../node_modules/nan/nan_maybe_43_inl.h: In function ‘Nan::Maybe<bool> Nan::ForceSet(v8::Local<v8::Object>, v8::Local<v8::Value>, v8::Local<v8::Value>, v8::PropertyAttribute)’:
../node_modules/nan/nan_maybe_43_inl.h:88:15: error: ‘class v8::Object’ has no member named ‘ForceSet’
   return obj->ForceSet(GetCurrentContext(), key, value, attribs);
               ^~~~~~~~
In file included from ../src/i2c.cc:3:
../node_modules/nan/nan.h: In function ‘v8::Local<v8::Value> Nan::MakeCallback(v8::Local<v8::Object>, v8::Local<v8::Function>, int, v8::Local<v8::Value>*)’:
../node_modules/nan/nan.h:817:60: warning: ‘v8::Local<v8::Value> node::MakeCallback(v8::Isolate*, v8::Local<v8::Object>, v8::Local<v8::Function>, int, v8::Local<v8::Value>*)’ is deprecated: Use MakeCallback(..., async_context) [-Wdeprecated-declarations]
         v8::Isolate::GetCurrent(), target, func, argc, argv);
                                                            ^
In file included from ../src/i2c.cc:1:
/usr/include/nodejs/src/node.h:177:50: note: declared here
                 NODE_EXTERN v8::Local<v8::Value> MakeCallback(
                                                  ^~~~~~~~~~~~
/usr/include/nodejs/src/node.h:91:42: note: in definition of macro ‘NODE_DEPRECATED’
     __attribute__((deprecated(message))) declarator
                                          ^~~~~~~~~~
In file included from ../src/i2c.cc:3:
../node_modules/nan/nan.h:817:60: warning: ‘v8::Local<v8::Value> node::MakeCallback(v8::Isolate*, v8::Local<v8::Object>, v8::Local<v8::Function>, int, v8::Local<v8::Value>*)’ is deprecated: Use MakeCallback(..., async_context) [-Wdeprecated-declarations]
         v8::Isolate::GetCurrent(), target, func, argc, argv);
                                                            ^
In file included from ../src/i2c.cc:1:
/usr/include/nodejs/src/node.h:177:50: note: declared here
                 NODE_EXTERN v8::Local<v8::Value> MakeCallback(
                                                  ^~~~~~~~~~~~
/usr/include/nodejs/src/node.h:91:42: note: in definition of macro ‘NODE_DEPRECATED’
     __attribute__((deprecated(message))) declarator
                                          ^~~~~~~~~~
In file included from ../src/i2c.cc:3:
../node_modules/nan/nan.h: In function ‘v8::Local<v8::Value> Nan::MakeCallback(v8::Local<v8::Object>, v8::Local<v8::String>, int, v8::Local<v8::Value>*)’:
../node_modules/nan/nan.h:831:62: warning: ‘v8::Local<v8::Value> node::MakeCallback(v8::Isolate*, v8::Local<v8::Object>, v8::Local<v8::String>, int, v8::Local<v8::Value>*)’ is deprecated: Use MakeCallback(..., async_context) [-Wdeprecated-declarations]
         v8::Isolate::GetCurrent(), target, symbol, argc, argv);
                                                              ^
In file included from ../src/i2c.cc:1:
/usr/include/nodejs/src/node.h:170:50: note: declared here
                 NODE_EXTERN v8::Local<v8::Value> MakeCallback(
                                                  ^~~~~~~~~~~~
/usr/include/nodejs/src/node.h:91:42: note: in definition of macro ‘NODE_DEPRECATED’
     __attribute__((deprecated(message))) declarator
                                          ^~~~~~~~~~
In file included from ../src/i2c.cc:3:
../node_modules/nan/nan.h:831:62: warning: ‘v8::Local<v8::Value> node::MakeCallback(v8::Isolate*, v8::Local<v8::Object>, v8::Local<v8::String>, int, v8::Local<v8::Value>*)’ is deprecated: Use MakeCallback(..., async_context) [-Wdeprecated-declarations]
         v8::Isolate::GetCurrent(), target, symbol, argc, argv);
                                                              ^
In file included from ../src/i2c.cc:1:
/usr/include/nodejs/src/node.h:170:50: note: declared here
                 NODE_EXTERN v8::Local<v8::Value> MakeCallback(
                                                  ^~~~~~~~~~~~
/usr/include/nodejs/src/node.h:91:42: note: in definition of macro ‘NODE_DEPRECATED’
     __attribute__((deprecated(message))) declarator
                                          ^~~~~~~~~~
In file included from ../src/i2c.cc:3:
../node_modules/nan/nan.h: In function ‘v8::Local<v8::Value> Nan::MakeCallback(v8::Local<v8::Object>, const char*, int, v8::Local<v8::Value>*)’:
../node_modules/nan/nan.h:845:62: warning: ‘v8::Local<v8::Value> node::MakeCallback(v8::Isolate*, v8::Local<v8::Object>, const char*, int, v8::Local<v8::Value>*)’ is deprecated: Use MakeCallback(..., async_context) [-Wdeprecated-declarations]
         v8::Isolate::GetCurrent(), target, method, argc, argv);
                                                              ^
In file included from ../src/i2c.cc:1:
/usr/include/nodejs/src/node.h:163:50: note: declared here
                 NODE_EXTERN v8::Local<v8::Value> MakeCallback(
                                                  ^~~~~~~~~~~~
/usr/include/nodejs/src/node.h:91:42: note: in definition of macro ‘NODE_DEPRECATED’
     __attribute__((deprecated(message))) declarator
[... 221 lines cut ...]
make: *** [i2c.target.mk:101: Release/obj.target/i2c/src/i2c.o] Error 1
[... cut ...]
gyp ERR! node -v v10.15.2
gyp ERR! node-gyp -v v3.8.0
gyp ERR! not ok 
npm WARN cedula@1.0.0 No repository field.

npm ERR! code ELIFECYCLE
npm ERR! errno 1
npm ERR! i2c@0.2.3 install: `node-gyp rebuild`
npm ERR! Exit status 1
npm ERR! 
npm ERR! Failed at the i2c@0.2.3 install script.
npm ERR! This is probably not a problem with npm. There is likely additional logging output above.

npm ERR! A complete log of this run can be found in:
npm ERR!     /home/pi/.npm/_logs/2019-09-24T22_41_16_622Z-debug.log

rzr, 2019-09-25:
As fallback try:

     npm install abandonware/i2c

jscaglia, 2019-09-25:
> As fallback try:
> 
> ```
>  npm install abandonware/i2c
> ```

Is it possible that the repository is private?

npm ERR! Error while executing:
npm ERR! /usr/bin/git ls-remote -h -t ssh://git@github.com/abandonware/i2c.git
npm ERR! 
npm ERR! Warning: Permanently added the RSA host key for IP address '140.82.113.4' to the list of known hosts.
npm ERR! git@github.com: Permission denied (publickey).
npm ERR! fatal: No se pudo leer del repositorio remoto.
npm ERR! 
npm ERR! Por favor asegúrese que tiene los permisos de acceso correctos
npm ERR! y que el repositorio existe.
npm ERR! 
npm ERR! exited with error code: 128

npm ERR! A complete log of this run can be found in:
npm ERR!     /home/pi/.npm/_logs/2019-09-25T12_46_29_019Z-debug.log

matteliot, 2019-09-25:
fwiw, I swapped to i2c-bus with minimal code changes and it's fine, I'm
using up to 12.4 or something:
https://www.npmjs.com/package/i2c-bus

jscaglia, 2019-09-25:
> fwiw, I swapped to i2c-bus with minimal code changes and it's fine, I'm using up to 12.4 or something: https://www.npmjs.com/package/i2c-bus
> […](#)

Hello, I tried to use this library but the problem is that is not compatible with npm PN532.
I have a raspberry pi 3 with an NFC board and I need to save data on an NFC card.
The PN532 library I see that it can do it without reaching such a low level.

Do you know if there is any way to do it with another library in case you can't with it?

http://wiki.sunfounder.cc/index.php?title=PN532_NFC_Module_for_Raspberry_Pi
https://www.npmjs.com/package/pn532

rzr, 2020-01-13:
Review welcome

Please test
https://www.npmjs.com/package/@abandonware/i2c/v/0.2.4-1

And confirm it did not break your applications