kelly/node-i2c #98: I2C does not work in the latest node version

jscaglia, 2019-09-24:
Hi, I tried to install de node module to mi nodejs app but I can't. I suppose that the reason is my nodejs and NPM version.
Do you know if the library is not compatible with the last version of nodeJS? If this is the case, do you know how I could install it?

Regards.

zvaehn, 2019-11-18:
I have tried the following and it worked for me (somehow):
1. `npm list -g | head -1` which gives you your global npm directory 
(in my case `/home/pi/.nvm/versions/node/v8.0.0/lib`)
2. Change your working dir to this directory (`cd ...`)
3. Try a regular `npm install i2c` which will install i2c and it's dependencies in the subfolder node_modules
4. In my case i was able to install the package depending on i2c without sudo and it worked.

shidevil, 2019-12-15:
Does it work on node12?

rzr, 2020-02-24:
Meanwhile linked patch is merged , you can use:

https://www.npmjs.com/package/@abandonware/i2c

jscaglia, 2020-05-27:
> I have tried the following and it worked for me (somehow):
>
> 1. `npm list -g | head -1` which gives you your global npm directory
>    (in my case `/home/pi/.nvm/versions/node/v8.0.0/lib`)
> 2. Change [...]

Unfortunately it didn't work for me. My global npm directory is '/usr/local/lib' and when I tried install with and without sudo print this.

pi@raspberrypi:/usr/local/lib $ sudo npm install i2c
npm WARN npm npm does not support Node.js v10.19.0
npm WARN npm You should probably upgrade to a newer version of node as we
npm WARN npm can't make any promises that npm will work with this version.
npm WARN npm Supported releases of Node.js are the latest release of 4, 6, 7, 8, 9.
npm WARN npm You can find the latest version at https://nodejs.org/
npm WARN deprecated coffee-script@1.9.1: CoffeeScript on NPM has moved to "coffeescript" (no hyphen)

> i2c@0.2.3 install /usr/local/lib/node_modules/i2c
> node-gyp rebuild

gyp ERR! configure error 
gyp ERR! stack Error: EACCES: permission denied, mkdir '/usr/local/lib/node_modules/i2c/build'
gyp ERR! System Linux 4.19.97-v7+
gyp ERR! command "/usr/bin/node" "/usr/bin/node-gyp" "rebuild"
gyp ERR! cwd /usr/local/lib/node_modules/i2c
gyp ERR! node -v v10.19.0
gyp ERR! node-gyp -v v3.8.0
gyp ERR! not ok 
npm ERR! code ELIFECYCLE
npm ERR! errno 1
npm ERR! i2c@0.2.3 install: `node-gyp rebuild`
npm ERR! Exit status 1
npm ERR! 
npm ERR! Failed at the i2c@0.2.3 install script.
npm ERR! This is probably not a problem with npm. There is likely additional logging output above.

npm ERR! A complete log of this run can be found in:
npm ERR!     /root/.npm/_logs/2020-05-27T12_32_55_247Z-debug.log

rzr, 2020-05-27:
Relate-to: https://github.com/kelly/node-i2c/pull/96