kelly/node-i2c #101: Segmentation fault on Scan. using @abandonware/i2c"

mitch85, 2020-06-05:
After update node to 12.x and update i2c to lates version of @abandonware/i2c. I got a segmentation fault on **new i2c();**

before with node v 8.x it works perfectly

hire-vladimir, 2020-07-03:
Also seeing **Segmentation fault** on `new i2c()` using node v12.18.1

```
pi@homebridge:~ $ node -v
v12.18.1
pi@homebridge:~ $ cat test.js
var i2c = require('@abandonware/i2c');
var address = 0x5c;
console.log("set address");
var wire = new i2c(address, {device: '/dev/i2c-1'}); 
console.log("init wire");
pi@homebridge:~ $ node test.js
set address
Segmentation fault
pi@homebridge:~ $
```

raniglas, 2020-07-07:
Same here - segmentation fault on new i2c() with node v11.15.0
```
var i2c = require('@abandonware/i2c');
var address = 0x21;
var wire = new i2c(address, {device: '/dev/i2c-1'}); 
Segmentation fault (core dumped)
```

This is on Raspberry Pi4 aarch64.

rzr, 2020-10-20:
Please confirm on latest release

Also comment welcome at:

https://github.com/kelly/node-i2c/issues/97